wiredopposite/OGX-Mini #5: Nintendo Switch output

Invictaz, 2024-03-03:
There isn't any real solution right now for Nintendo Switch unhacked consoles.

We have sys-con but it requires a hacked console.

https://github.com/cathery/sys-con

And we have https://github.com/gdsports/NSGadget_Teensy

But it requires a Teensy. Not a much cheaper Pico.

There is

https://github.com/juan518munoz/PicoSwitch-WirelessGamepadAdapter

But it is bluetooth and not usb input

I hope you can use these projects and the Dekunukem documentation to have Nintendo Switch output.

ImSammy6, 2024-03-11:
The description says it supports switch, does it not work for you?

Invictaz, 2024-03-11:
> ~The description says it supports switch, does it not work for you?~
> 
> Actually it seems like that was added after you posted this

I understand your question. However, the 0.2.0 version added various Switch controllers, not changed the output.
It depends on the game how the adapter should output as not all Switch games accept Pro Controller input.

Maybe it is needed to emulate 2 joycons. Therefore I would like to keep the discussion open on Switch output.

Invictaz, 2024-03-12:
I double checked. The emulated Switch controller is not a pro controller so it cannot be remapped in the Switch menu.
See https://github.com/jfedor2/hid-remapper/issues/118 for screenshots.

The 8Bitdo adapter 2 connected directly to a Switch does emulate a Switch Pro, however if you connect a 8bitdo Pro 2 via the OGX-Mini it cannot be in Switch output mode (view+LB). Only in right + view mode. Not sure why.

wiredopposite, 2024-03-12:
I honestly don't know that much about the joycons and which games require them, I'm not even sure they're connectable via USB. If not then emulating them is not an option for the Pico. I also don't know how the 8bitdo behaves when switching modes so it could get handled incorrectly if it's not in Xinput mode, I'll add that to the readme.

Both the [NSGadget_Teensy](https://github.com/gdsports/NSGadget_Teensy) and [PicoSwitch-WirelessGamepadAdapter](https://github.com/juan518munoz/PicoSwitch-WirelessGamepadAdapter) are doing the same thing as the OGX-Mini, they're emulating a basic wired non-pro gamepad. I do plan on improving Switch support (similar to 8bitdo) after I iron out some issues with PS3, I have a hardware USB sniffer coming soon so that'll make it way easier for me.

Invictaz, 2024-03-12:
The joycons support usb when connected to this handle

[image]

I will reply to the other questions soon

Invictaz, 2024-03-17:
There are some games that don't support the Switch Pro. Most known one is Super Mario Party.

1) Super Mario Party

2) The World Ends with You

3) Ring Fit

4) Nintendo Labo

5) Just Dance Series

6) 1-2-Switch

7) Pokémon: Let's Go, Pikachu! and Let's Go, Eevee!

8) The Stretchers (Doesn't support Multiplayer mode)

9) Clubhouse Games: 51 Worldwide Classics (Doesn't support Motion-Based Games EXP: darts, bowling)

10) Resident Evil Revelations 1 & 2 (Doesn't support Gyro aiming)

11) Astral Chain (Doesn't support Multiplayer Mode)

12) SEGA AGES Virtua Racing (Doesn't support Multiplayer mode)

13) Fitness Boxing 1 and 2

14) Jump Rope Challenge

Invictaz, 2024-03-24:
Here is the output of the 8bitdo pro 2 adapter in Nintendo Switch mode (Select + L1)

LSUSB -d

`ubuntu@ubuntu:~/Downloads/xone$ lsusb -d 057e:2009
Bus 001 Device 027: ID 057e:2009 Nintendo Co., Ltd Pro Controller
`
DMESG

```
 8781.416949] i915 0000:00:02.0: [drm] HPD interrupt storm detected on connector HDMI-A-3: switching from hotplug detection to polling
[12279.162834] usb 1-4: USB disconnect, device number 18
[12294.548277] usb 1-3: USB disconnect, device number 3
[12302.632451] usb 1-4: new full-speed USB device number 19 using xhci_hcd
[12302.782576] usb 1-4: New USB device found, idVendor=2dc8, idProduct=3106, bcdDevice= 1.00
[12302.782588] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[12302.782594] usb 1-4: Product: 8BitDo Receiver
[12302.782599] usb 1-4: Manufacturer: 8BitDo
[12302.782603] usb 1-4: SerialNumber: E417D8743AB1
[12303.181994] usb 1-4: USB disconnect, device number 19
[12303.488452] usb 1-4: new full-speed USB device number 20 using xhci_hcd
[12303.638534] usb 1-4: New USB device found, idVendor=045e, idProduct=028e, bcdDevice= 1.14
[12303.638546] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[12303.638552] usb 1-4: Product: Controller
[12303.638556] usb 1-4: Manufacturer: 8BitDo
[12303.638561] usb 1-4: SerialNumber: E417D8743AB1
[12304.039159] usb 1-4: USB disconnect, device number 20
[12304.352471] usb 1-4: new full-speed USB device number 21 using xhci_hcd
[12304.502425] usb 1-4: New USB device found, idVendor=057e, idProduct=2009, bcdDevice= 2.10
[12304.502437] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[12304.502443] usb 1-4: Product: Pro Controller
[12304.502448] usb 1-4: Manufacturer: Nintendo Co., Ltd.
[12304.502452] usb 1-4: SerialNumber: E417D8743AB1
[12304.506169] input: Nintendo Co., Ltd. Pro Controller as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:057E:2009.0026/input/input66
[12304.506815] hid-generic 0003:057E:2009.0026: input,hidraw8: USB HID v1.11 Joystick [Nintendo Co., Ltd. Pro Controller] on usb-0000:00:14.0-4/input0
[12305.102230] usb 1-4: USB disconnect, device number 21
[12305.428495] usb 1-4: new full-speed USB device number 22 using xhci_hcd
[12305.577463] usb 1-4: New USB device found, idVendor=054c, idProduct=0cda, bcdDevice= 1.00
[12305.577476] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[12305.577482] usb 1-4: Product: Controller
[12305.577486] usb 1-4: Manufacturer: Sony Interactive Entertainment
[12305.579793] input: Sony Interactive Entertainment Controller as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:054C:0CDA.0027/input/input67
[12305.580243] hid-generic 0003:054C:0CDA.0027: input,hidraw8: USB HID v1.11 Gamepad [Sony Interactive Entertainment Controller] on usb-0000:00:14.0-4/input0
[12305.581349] input: Sony Interactive Entertainment Controller Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.1/0003:054C:0CDA.0028/input/input68
[12305.640785] input: Sony Interactive Entertainment Controller System Control as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.1/0003:054C:0CDA.0028/input/input69
[12305.641143] hid-generic 0003:054C:0CDA.0028: input,hidraw9: USB HID v1.10 Device [Sony Interactive Entertainment Controller] on usb-0000:00:14.0-4/input1
[12306.574867] usb 1-4: USB disconnect, device number 22
[12306.956520] usb 1-4: new full-speed USB device number 23 using xhci_hcd
[12307.106505] usb 1-4: New USB device found, idVendor=057e, idProduct=2009, bcdDevice= 2.10
[12307.106517] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[12307.106523] usb 1-4: Product: Pro Controller
[12307.106528] usb 1-4: Manufacturer: Nintendo Co., Ltd.
[12307.106532] usb 1-4: SerialNumber: E417D8743AB1
[12307.110058] input: Nintendo Co., Ltd. Pro Controller as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:057E:2009.0029/input/input70
[12307.168950] hid-generic 0003:057E:2009.0029: input,hidraw8: USB HID v1.11 Joystick [Nintendo Co., Ltd. Pro Controller] on usb-0000:00:14.0-4/input0
[12308.301745] usb 1-4: USB disconnect, device number 23
[12308.624545] usb 1-4: new full-speed USB device number 24 using xhci_hcd
[12308.774362] usb 1-4: New USB device found, idVendor=2dc8, idProduct=3107, bcdDevice= 2.00
[12308.774374] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[12308.774380] usb 1-4: Product: IDLE
[12308.774385] usb 1-4: Manufacturer: 8BitDo
[12308.774389] usb 1-4: SerialNumber: E417D8743AB1
[12308.777847] hid-generic 0003:2DC8:3107.002A: hiddev1,hidraw8: USB HID v1.10 Device [8BitDo IDLE] on usb-0000:00:14.0-4/input0
[12324.914550] usb 1-4: USB disconnect, device number 24
[12325.372795] usb 1-4: new full-speed USB device number 25 using xhci_hcd
[12325.522496] usb 1-4: New USB device found, idVendor=2dc8, idProduct=3107, bcdDevice= 2.00
[12325.522508] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[12325.522515] usb 1-4: Product: IDLE
[12325.522519] usb 1-4: Manufacturer: 8BitDo
[12325.522523] usb 1-4: SerialNumber: E417D8743AB1
[12325.525653] hid-generic 0003:2DC8:3107.002B: hiddev1,hidraw8: USB HID v1.10 Device [8BitDo IDLE] on usb-0000:00:14.0-4/input0
[12327.556372] usb 1-4: USB disconnect, device number 25
[12328.028819] usb 1-4: new full-speed USB device number 26 using xhci_hcd
[12328.179225] usb 1-4: New USB device found, idVendor=2dc8, idProduct=3106, bcdDevice= 1.00
[12328.179237] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[... cut ...]
[12334.471262] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[12334.471269] usb 1-4: Product: Pro Controller
[12334.471273] usb 1-4: Manufacturer: Nintendo Co., Ltd.
[12334.471278] usb 1-4: SerialNumber: E417D8743AB1
[12334.477531] input: Nintendo Co., Ltd. Pro Controller as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:057E:2009.002C/input/input71
[12334.541466] hid-generic 0003:057E:2009.002C: input,hidraw8: USB HID v1.11 Joystick [Nintendo Co., Ltd. Pro Controller] on usb-0000:00:14.0-4/input0
```

Invictaz, 2024-03-24:
[dump.txt](https://github.com/wiredopposite/OGX-Mini/files/14734246/dump.txt)