wiredopposite/OGX-Mini #6: Question: BlueRetro support?

MrKittenK, 2024-03-07:
I was wondering if this has support for BlueRetro? 
There is a GitHub, I've linked, that sends commands from BlueRetro to ogx360 over I2C

https://github.com/konwektor/BlueRetro

Could this work/be Intergrated? 

Thank you for the great work so far ☺️

Invictaz, 2024-03-10:
@MrKittenK So you want to have Blueretro working on a Pico or use a Blueretro adapter on the usb input side of this OGX adapter? Then you will have two adapters in a row?

bbsan2k, 2024-03-30:
How about adding BluePad32 library to be used with a Pico W? It could be used as an alternative to USB Host if no usb dev is detected…

Invictaz, 2024-03-30:
@bbsan2k It would be wise to see what the pros and cons are. Bluepad32 is already ported to RP2040, see the PicoSwitch github, but Blueretro seems to support more controllers, has longer development and has different features. It's difficult to say.

bbsan2k, 2024-03-31:
It probably comes down to how much time you want to invest into that. BlueRetro IIRC has lots of dependencies into the ESP-IDF which first have to stubbed using the Pico SDK. Additionally it uses more resources.
BluePad is more or less Plug and Play and can be integrated in way less time. I‘ll probably also have  a deeper look in integrating BluePad in the coming weeks, if I find some spare time😉

Invictaz, 2024-03-31:
So in essence a Frankenstein mod of PicoSwitch and OGX Mini would happen. Not sure about the ESP32 stuff. I think it is possible to port Blueretro to Pico but resources are limited on both platforms. Especially the WebUI is something that Bluepad32 doesn't have. The webUI makes controller configuration and remapping possible.

Invictaz, 2024-12-20:
Bluetooth is now added in v1.0 alpha so possibly this can be closed @MrKittenK