wiredopposite/OGX-Mini #7: MPOW Wired Gamepad Device ID

CloudyShane, 2024-03-09:
I own a chinese X-Input gamepad, but it has a catch. It was developed to be used with a damn PS4... So it only works in X-Input on a PC. On other devices it refuses to work. It never worked on a RGH Xbox 360 (with a gamepad patch) and on OGX Mini.z

It's called MPOW, and it's a wired one (there is a D-Input only wireless version available aswell).

Here are it's IDs:
USB\VID_045E&PID_028E&REV_0110
USB\VID_045E&PID_028E

wiredopposite, 2024-03-09:
0x045E and 0x028E are IDs for an Xbox 360 controller, so it's probably changing its descriptors to show up as XInput after it's mounted, I'm not sure if there's some command that has to be sent to the gamepad to make it do that though. 

You can install Wireshark with USBPCap, monitor your USB devices and then plug the gamepad into your PC, wait a second and then click the stop button, that will show you all the packets being sent and received, if the descriptors change later on, or if it's actually just using Xbox controller IDs. In the Info column, it'll say "GET DESCRIPTOR Response DEVICE" for the relevant USB packet.

I am planning changing how different types of gamepads are detected eventually, so maybe that will fix this issue because there are plenty of third party ones that do the same thing.

CloudyShane, 2024-03-09:
That was an interesting read!
I don't know anything about wireshark, but I've managed to capture the data you asked for. I will pin it below:
[mpow.zip](https://github.com/wiredopposite/OGX-Mini/files/14547286/mpow.zip)
There is the _.pcapng_ file inside with all the handshaking. Hope it helps!

CloudyShane, 2024-03-09:
I've read into the report, and yes, it starts out as a "Wired PS4 Controller", and after some time changes itself into Microsoft Xbox 360 Controller, just like you've said.

CloudyShane, 2024-03-09:
The original VID and PID are 0x2563 and 0x0357 as it seems.

wiredopposite, 2024-03-10:
Thanks for that, I added those IDs to the list for PS4 so hopefully it'll work. I'm still not 100% on how the gamepad knows to change descriptors like that, but anyway, it'll be included in the next update probably coming tomorrow.

CloudyShane, 2024-03-10:
>  it'll be included in the next update probably coming tomorrow.

Thank you! I will test the update as soon as I can and will report my results here.

wiredopposite, 2024-03-12:
Okay I've published [v0.3.0](https://github.com/wiredopposite/OGX-Mini/releases/tag/v0.3.0), let me know if that works for your controller