wiredopposite/OGX-Mini #10: Playstation 4 output

Invictaz, 2024-03-14:
There is quite some HID report information available. The only problem is the 8 minute timeout

https://controllers.fandom.com/wiki/Game_Controller_Collective_Wiki

Invictaz, 2024-03-24:
Seems Santroller has HID output on PS4

https://github.com/Santroller/Santroller/blob/16d86b1004f3f73739e5b723e38c98126b578642/src/shared/hid/hid.cpp#L41

Invictaz, 2024-04-17:
Another recent authentication implementation

https://github.com/jfedor2/paaas

Invictaz, 2024-12-23:
See #63 for additional info.

mrdude2478, 2024-12-29:
> There is quite some HID report information available. The only problem is the 8 minute timeout
> 
> https://controllers.fandom.com/wiki/Game_Controller_Collective_Wiki

Maybe something from here can help with that:
https://drive.google.com/file/d/1CrmvLvmQDPeA_14ausDpVmS5BRlEc19A/view