wiredopposite/OGX-Mini #11: WIIU Output

Invictaz, 2024-03-16:
It seems that the Brook converter (not longer available for purchase) mimics a WiiMote over USB and the WiiU accepts that. Maybe this was because of a development kit function where the Wiimotes were wired instead of wireless.

https://hackinformer.com/2017/07/23/super-converter-nintendo-switch-wii-u/

A wiimote can be emulated. Will post the information and code how to do that.

wiredopposite, 2024-03-17:
Does the WiiU have some kind of controller security protocol? I don't own one

Invictaz, 2024-03-22:
@wiredopposite I took some time to prepare your question. Yes there is security, but the protocol is made opensource by Marcan and recently imported into Blueretro.

An Arduino library is here
https://gist.github.com/sm00th/d1148f11692cb9fb62cc

It emulates a Wii Classic Controller connected as an extension to a Wiimote. 

You can also try to emulate a Pokken controller, which works both on WiiU and on Switch, but on the WiiU apparently only in specific games. It's one of the only controllers that I know of that have USB input on the WiiU.

https://github.com/Jas2o/Leonardo-Switch-Controller

The current situation on the WiiU is as follows:

We have HID2Vpad on a plugin for Aroma (a homebrew variant) but it requires an .ini configuration 
Some demo files here: https://github.com/Maschell/controller_patcher_configs
Originally it was only usable under Tiramisu (the older homebrew environment) 
https://github.com/Maschell/hid_to_vpad

The main problem with HID2Vpad is that it doesn't allow a lot of controllers. The usuability of HID controllers seems limited on what Nintendo seems to support and this is not fully known or explored.

We also have Bloopair which emulates a WiiU pro controller. This would be a good situation since quite a lot of WiiU games support the pro controller. 

https://github.com/GaryOderNichts/Bloopair/blob/main/ios/ios_pad/source/controllers.c

However, the bluetooth chip on the WiiU is old and it doesn't allow BLE controllers like the Raspberry Pico does. 

On Wii the bluetooth chip is even older, and there is almost no support for controllers only in some homebrew, for a wired Playstation 3 controller. Systemwide there is no solution yet. We have Fakemote, but it only works in homebrew and emulates a Wiimote over usb. Fakemote does not work systemwide. 

https://github.com/xerpi/fakemote/blob/main/source/fake_wiimote.c

We also have GC2Wiimote but it only works on heavily portabilized Wii consoles AFAIK. It allows Wii games to be played with an emulated Wii controller with gamecube controller as input (which they use on portable consoles). 

https://github.com/search?q=repo%3AAurelio92%2FRVLoader%20gc2wiimote&type=code

The best solution would be to have both a WiiU Pro Controller emulated over USB and a Wiimote, so it can be natively used without homebrew and additional hacks. Since a lot of code is already there it should be doable to port something to the Pico side. 

If you have questions let me know.