wiredopposite/OGX-Mini #12: Feature Request: Headset Support

jamesah2008, 2024-03-19:
LOVE that you made this. It's SO easy to get an Xbox One controller working. EPIC.

Just wondering what it would take to make voice chat work through a controller so we can use it on the OG Xbox. I know the OEM Microsoft Xbox Communicator is just a device that plugs into the Original S-controller memory card slot.

I don't have the skills but I'd be super curious to know what it would take to add one of the following:

1. Xbox One wired controller support
2. Bluetooth voice support that imitates the OG Xbox Communicator (or just use a Hub that?)

parsonsalan1, 2024-04-12:
yes please

Invictaz, 2024-04-12:
@jamesah2008 you need the Xbox 1790 stick for headset audio over bluetooth see #14 

For OG Xbox it is unknown what is needed.

jamesah2008, 2024-04-15:
@Invictaz There's been a good bit of discussion on the XBOX Dev discord regarding dumping the firmware from the OG Communicator. https://github.com/Ryzee119/Xblc_Dumper

Don't know what it truly takes but I'm fascinated to watch the developments. Maybe I can contribute documentation or something.

I would love to see that 1790 stick work with the OGX-Mini. I don't have a wireless adapter yet and was thinking of making that my first purchase.

Invictaz, 2024-04-15:
@jamesah2008 you happen to have a link for that discord? I'm fascinated by peripheral investigation and hardware dumping and reverse engineering.

jamesah2008, 2024-04-15:
https://discord.com/channels/428359196719972353/428392158555406336/428394009204097025

jamesah2008, 2024-04-15:
> @jamesah2008 you happen to have a link for that discord? I'm fascinated by peripheral investigation and hardware dumping and reverse engineering.

https://discord.com/channels/428359196719972353/428392158555406336/428394009204097025

Invictaz, 2024-04-15:
@jamesah2008 I believe the UAC chip is also in my Reloop DJ mixer as USB audio card. Not really sure but I can check. Thanks for the link.

Invictaz, 2024-04-27:
@jamesah2008 see this new repo.

https://github.com/Ryzee119/hawk

Maybe some code can be used in this project.

jamesah2008, 2024-04-27:
😍

wiredopposite, 2024-12-09:
So I've looked into this and it would be extremely difficult for me to do with just a Pico and my programming experience. It would need some bare metal programming, TinyUSB is not capable of this since the original Xbox controller is not a composite device, it's actually a USB "hub" and the microphone is its own separate USB device with its own root hub port.

A possible way to make this work is to use a USB splitter (I haven't tried this), plug the OGX-Mini into one port and an Xbox communicator into another. By splitter I don't mean a hub, as far as I know normal hubs can't be enumerated by the Xbox. Another way would be to modify the Xbox's USB stack somehow but that's a tougher one.

mastercheff, 2024-12-09:
Hey I do play Halo CE cometetively too, lag is under 1 ms at most. With fast controller (xbox polls at 250hz so You get benefit of using non-MS equipement) its even quicker than the S controller

jamesah2008, 2024-12-09:
Thanks for the tip! What controller should I use? Is there a way to read
the polling rate?

James

mastercheff, 2024-12-09:
For measuring both Duke and S controller , I used OGXmini in Xinput mode and this app : 

https://github.com/chrizonix/XInputTest

(Duke came out on top 250hz, S ~150hz)

For lan games we use 8bitdo Ultimate 2C , Gamesir Nova pads they're confirmed fast and affordable

jamesah2008, 2024-12-09:
Awesome, thanks for the info!