wiredopposite/OGX-Mini #13: Feature Request: "memory card" feature

jamesah2008, 2024-03-19:
I would love to be able to use this (with a controller hooked in) as a memory card to perform the end-game exploit. This is occurring to me because I don't have another original xbox dongle handy or two controllers.

A function like this would round out emulating all the functions of the original xbox controller.

wiredopposite, 2024-03-22:
I would like to do that and am looking into it, but I can't promise it at this point and can pretty much guarantee it would require one of those 16MB Picos if anything. Another option would be soldering an SD card slot to the Pico and passing it through that way.

Invictaz, 2024-03-22:
There is 

https://github.com/zwenergy/CPAK2040

and 

https://github.com/dangiu/PicoMemcard

and 

SD2PSX

https://github.com/bbsan2k/sd2psx_firmware

The problem is that quite some of these projects eat up a lot of RAM. 

The ideal project would be Blueretro + Save storage but there are limits to what these microcontrollers can do.

Invictaz, 2024-12-20:
Maybe now there is Pico 2 this can be achieved