wiredopposite/OGX-Mini #14: Support Xbox 1790 adapter

Invictaz, 2024-03-24:
This adapter allows Xbox GIP which is a much lower latency than over bluetooth. It's a wifi protocol working over 5GHZ.

DMESG 

`[ 3907.536732] usb 1-4: USB disconnect, device number 16
[ 3912.380094] usb 1-4: new high-speed USB device number 17 using xhci_hcd
[ 3912.529455] usb 1-4: New USB device found, idVendor=045e, idProduct=02fe, bcdDevice= 1.00
[ 3912.529467] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 3912.529474] usb 1-4: Product: XBOX ACC
[ 3912.529478] usb 1-4: Manufacturer: Microsoft Inc.
[ 3912.529483] usb 1-4: SerialNumber: 023620

LSUSB

`ubuntu@ubuntu:~/Downloads$ sudo lsusb -vd 045e:02fe

Bus 001 Device 017: ID 045e:02fe Microsoft Corp. XBOX ACC
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.01
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x045e Microsoft Corp.
  idProduct          0x02fe 
  bcdDevice            1.00
  iManufacturer           1 Microsoft Inc.
  iProduct                2 XBOX ACC
  iSerial                 3 023620
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x004a
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          2 XBOX ACC
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           8
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              2 XBOX ACC
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x85  EP 5 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x08  EP 8 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x05  EP 5 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x06  EP 6 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x07  EP 7 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x09  EP 9 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength       0x000c
  bNumDeviceCaps          1
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x0000f41e
      BESL Link Power Management (LPM) Supported
    BESL value     1024 us 
    Deep BESL value    61440 us 
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0002
  (Bus Powered)
  Remote Wakeup Enabled
`

https://github.com/medusalix/xone
https://github.com/paroj/xpad/issues/25
http://web.archive.org/web/20190218065058/https://gist.github.com/devkid/4b3bd50760504d1b93ea684cfd3ed895
https://solus.pkgs.org/rolling/solus-shannon-x86_64/xone-0.3-27-1-x86_64.eopkg.html

EVDEV-Joystick

ubuntu@ubuntu:~/Downloads$ evdev-joystick --l
/dev/input/by-id/usb-Microsoft_Inc._XBOX_ACC_023620-event-joystick

`ubuntu@ubuntu:~/Downloads$ evdev-joystick --s /dev/input/by-id/usb-Microsoft_Inc._XBOX_ACC_023620-event-joystick
Supported Absolute axes:
  Absolute axis 0x00 (0) (X Axis) (value: 1827, min: -32768, max: 32767, flatness: 128 (=0.39%), fuzz: 16)
  Absolute axis 0x01 (1) (Y Axis) (value: -3147, min: -32768, max: 32767, flatness: 128 (=0.39%), fuzz: 16)
  Absolute axis 0x02 (2) (Z Axis) (value: 0, min: 0, max: 1023, flatness: 0 (=0.00%), fuzz: 0)
  Absolute axis 0x03 (3) (X Rate Axis) (value: 2652, min: -32768, max: 32767, flatness: 128 (=0.39%), fuzz: 16)
  Absolute axis 0x04 (4) (Y Rate Axis) (value: 435, min: -32768, max: 32767, flatness: 128 (=0.39%), fuzz: 16)
  Absolute axis 0x05 (5) (Z Rate Axis) (value: 0, min: 0, max: 1023, flatness: 0 (=0.00%), fuzz: 0)
  Absolute axis 0x10 (16) (Hat zero, x axis) (value: 0, min: -1, max: 1, flatness: 0 (=0.00%), fuzz: 0)
  Absolute axis 0x11 (17) (Hat zero, y axis) (value: 0, min: -1, max: 1, flatness: 0 (=0.00%), fuzz: 0)
`

Invictaz, 2024-03-24:
Joystick test values

[evtest-joystick.txt](https://github.com/wiredopposite/OGX-Mini/files/14733992/evtest-joystick.txt)

wiredopposite, 2024-12-15:
This one can happen but I've prioritized getting Pico W firmware ironed out lately, the official Xbox wireless adapter doesn't work like a normal wireless dongle, I'm pretty sure it gets its firmware from the computer it's plugged into. 

A way to do this would be like how the xpadneo and xone drivers do it, I'd have to compile the adapter's firmware in with the uf2 file for the pico and then feed the firmware to the adapter, which I don't think is actually all that complicated but I just haven't had time to sit down and come up with a driver that can do it.

Invictaz, 2024-12-16:
The firmware is in one of the links I posted above. If you do Pico W I hope you can compile it as well for Pico 2 W.