wiredopposite/OGX-Mini #15: Xbox 360 output

Invictaz, 2024-03-24:
Controller connection is encrypted, however the key can be retrieved.

https://github.com/GoobyCorp/Xbox-360-Crypto/blob/master/xsm3.py

Invictaz, 2024-10-01:
GP2040-CE just added Xbox 360 output mode with passthrough authentication

https://github.com/OpenStickCommunity/GP2040-CE/releases/tag/v0.7.10-RC3

Maybe this is useful

Invictaz, 2024-12-20:
Proper Xbox 360 output is something different than emulated Xbox 360 output as the real Xbox360 output also contains the keypad entries (if a keypad is connected).

barrenechea, 2025-05-06:
Plug and play authentication has been achieved by Santroller just a few hours ago https://github.com/Santroller/Santroller/commit/5b6d4aed090618d9e3645e1dddee366ec0d38803

Retail keys have also been ported to libxsm3 via https://github.com/InvoxiPlayGames/libxsm3/pull/1

Invictaz, 2025-05-06:
> Plug and play authentication has been achieved by Santroller just a few hours ago [Santroller/Santroller@5b6d4ae](https://github.com/Santroller/Santroller/commit/5b6d4aed090618d9e3645e1dddee366ec0d38803)
> 
> Retail keys have also been ported to libxsm3 via [InvoxiPlayGames/libxsm3#1](https://github.com/InvoxiPlayGames/libxsm3/pull/1)

Great news. If it works then we won't need an auth device or jailbroken Xbox360.
I hope @wiredopposite is still working on this fantastic software

MegaCadeDev, 2026-02-27:
> [@MegaCadeDev](https://github.com/MegaCadeDev) when the updates/fixes are merged onto this repo issue can be closed. Until then, the unofficial fork is here:
> 
> https://github.com/MegaCadeDev/OGX-Mini-2026

When the original guy comes back I'll push my fixes to him as well! Just want to contribute to this awesome project is all!