wiredopposite/OGX-Mini #16: Not outputting as original USB descriptor

Invictaz, 2024-03-30:
The OGX Mini doesn't use the original output descriptors which is problematic on some systems.

[image]

[image]

Invictaz, 2024-03-30:
Might be related to 

https://github.com/wiredopposite/OGX-Mini/issues/5#issuecomment-1989718179

wiredopposite, 2024-04-18:
The Vita TV thing is tough, I'm currently emulating a generic DInput controller for PS3 since true Dualshock 3 emulation has been a tough one to figure out. I have been working on it, I'm not sure if you made this after v0.3 but is this still an issue with Vita?

The Switch problem is easier to address so I'll work on that as well, can you tell me which software you're using with it?

Invictaz, 2024-04-30:
@wiredopposite There is a new project that emulates a Dualshock 3. Maybe you can have a look at it to see what code you can derive to have it working. 

https://github.com/isage/vivid

The screenshot you saw was on Windows but it is the same on the Switch. As it is not advertised as a real pro controller, there is no remap option available in the Nintendo Switch controller menu, making it much harder to work with since you have to recompile it to have a remap. I did recompile the PicoSwitch project with a remap, but it's quite a hassle to setup the SDK just to change some buttons. Especially if you use A as B on Nintendo and the other modes on another way.

Another (quick solution) would be to have the A and B button swap hidden under a key combo. I might suggest that in a new issue.

Invictaz, 2025-01-11:
So the descriptor is here

HPAC is according to @jfedor a unique ID that represents how many receivers are connected

https://github.com/wiredopposite/OGX-Mini/blob/b3bcbff50a713a0aaa87b42cf132132d8488bd58/Firmware/RP2040/src/USBDevice/tud_callbacks.cpp#L36