wiredopposite/OGX-Mini #21: Xbox one controller works for 30 seconds..

parsonsalan1, 2024-04-15:
Hello,

Using an xbox one controller, the input works great for about 20 seconds or so, then you start to walk slow and then you just spin in circles and vibration stays on (if you shoot a rocket or something).

Invictaz, 2024-04-16:
Which model number? 1537, 1608 or other?

parsonsalan1, 2024-04-16:
M1106909-007  

Thanks!

parsonsalan1, 2024-04-16:
Is there anything I can provide that would help? Should I try a different controller? Why would it do that?

wiredopposite, 2024-04-18:
I use that same model controller to test a whole bunch of stuff and don't have this issue. There's a couple of things you could try, one is using a different cable, if the cable isn't damaged you could also try one that has a thicker gauge wire, the controller may be having trouble getting enough power. You could also try updating the firmware through Xbox Accessories app in Windows. Does the controller have issues when you use it wired with something else like a PC?

closed 2024-04-30 by wiredopposite