wiredopposite/OGX-Mini #22: Input not working

Anpert70, 2024-04-19:
Hi!
So I've been trying to make this work with a PowerA Controller. The problem is that when connected, the controller does turn on the led onboard, but when I try pressing any button it gives no response.

wiredopposite, 2024-04-19:
Hi, I need some more info, what console is the PowerA controller meant for and which model is it?

Anpert70, 2024-04-19:
It is a Switch wired controller, and the model is 1521474-01

wiredopposite, 2024-04-19:
Okay I'll take a look tomorrow and see if I can reproduce the issue. Is it still a problem if you use an earlier version of the firmware?

Anpert70, 2024-04-19:
Thanks! And yeah, I've tried with older releases but it still isn't working.

wiredopposite, 2024-04-19:
Not sure why I blanked on this, but it's probably because I don't have that device's VID and PID included in the firmware. Can you plug it into a PC and post those? You can get them in Windows from Device Manager, right click the device (should be under human interface devices), select properties > details tab > hardware ids from the drop down. It'll look like this: HID\VID_046D&PID_C05A

Anpert70, 2024-04-19:
Sure! USB\VID_20D6&PID_A713

Invictaz, 2024-04-19:
@wiredopposite is it as simple as just adding PID and VID and recompiling or do we need more to add controller support?

If it is simple I might add some and do a pull request.

Anpert70, 2024-04-19:
Yup, tried it and it works now! Thanks!

closed 2024-04-19 by Anpert70