wiredopposite/OGX-Mini #23: FYI: This works on a modded Xbox 360

baabo, 2024-04-23:
I just tested this on an RGH modded Xbox 360 with [UsbdSecPatch](https://github.com/InvoxiPlayGames/UsbdSecPatch) installed. The adapter works without issues.
Thank you for creating this project!

Invictaz, 2024-04-23:
Interesting find.

Maybe we can add the 

"WgcAddDevice to skip over a check for a certain interface that is usually missing on certain custom controller firmwares."

To the OGX- mini ?

IzWhatitizz, 2024-04-23:
I tested this also. Tested Need for Speed the Run and analog sticks didn't work. All other buttons were fine.

baabo, 2024-04-23:
The analog sticks definitely worked for me when using a DualSense controller.

IzWhatitizz, 2024-04-24:
You're right there is something going on with my UsbdSecPatch plugin. Tried 4 different controllers and same result. Even tried 8bitdo v2 receiver, so it has to be my plugin.

baabo, 2024-04-24:
You could try updating the plugin to 1.1 version if it is not already on that version.
I also use the XeBuild patch and not the Dashlaunch patch. I like it better integrated into XeBuild as with that you don't have to replug the controller every time the console boots.

IzWhatitizz, 2024-04-24:
That's what I'm gonna have to do. I'm pretty sure that is my problem. Thx

baabo, 2024-06-02:
I now noticed that the left stick doesn't work in Aurora or Microsoft guide menu for browsing the menus. With a regular controller or another Xinput supporting adapter it works. I can imagine this could cause issues in some games, like @IzWhatitizz had.

lopenator, 2024-07-18:
this is actually working. amazing!

willsy32, 2024-07-24:
Also, OG Xbox controllers work as well, so you can finally use original controllers for backwards compatible games!

Invictaz, 2024-07-25:
I hope the fix by @pcm720 gets merged

Invictaz, 2024-09-09:
Seems that the 8bitdo Pro 2 adapter has the same problem in Xinput mode (no analog in dashboard, works fine in game Forza 2). So it might have something to do with the usb patch as well.

pcm720, 2024-09-09:
@Invictaz 
> Seems that the 8bitdo Pro 2 adapter has the same problem in Xinput mode (no analog in dashboard, works fine in game Forza 2). So it might have something to do with the usb patch as well.

Their USB Adapter 2 behaves the same way.
I've reported it to them back in July, and received a generic "we will give feedback to our team" in response.
Obviously, they did nothing about it, at least for now.

I'm also having issues using their Ultimate 3-mode Controller for Xbox. Seems that TinyUSB doesn't like how its receiver initializes and simply keeps resetting the bus. This prevents receiver from working properly, it can't even connect to the controller.
For now, I have to use Mayflash Magic NS2 on Xbox 360 and Mayflash Magic NS2+OGX-Mini combo on OG Xbox.

Invictaz, 2024-09-09:
For OG Xbox there is now a Blueretro dongle by Darthcloud64. If OGX Mini implements Bluepad32 or Blueretro this is not needed anymore.

Strange thing is the 8bitdo Adapter resets the output randomly giving warnings on screen that no controller is connected. 

Since the authentication is disabled with the usbSecPatch there must be a polling issue?

The Cronus Zen I have does not have Xbox 360 output but my Cronus Max does. I have to see how it works since on the Dualsense there might be some vibration issues (only left motor?) While the Xbox Elite 2 seems to work fine, but has very heavy vibration. I hope the upcoming webUI let's you setup which vibration motors are used, the trigger motors or small or big motors or on the Dualsense the haptics?

pcm720, 2024-09-09:
@Invictaz 

I wish it was that easy. Bluetooth mode on Ultimate 3-mode Controller is a useless DirectInput mode that doesn't even support vibration and can't connect to BlueRetro for some reason, so the only way to use it properly is to use the receiver 😞 

I didn't test with anything other than 1.1 version of the patch. Didn't see any disconnect issues with the USB Wireless Adapter 2, but I haven't really tested it for more than ten minutes.

IzWhatitizz, 2024-09-09:
I noticed with my 8bitdo v2 that I had voltage drops from the rp2040. I ran a wire to bypass to the usb host just to make sure that the 8bitdo had steady voltage.  Running through the rp2040 it would drop from 5 to 3.7 volts. This has improved reliability in connection.

baabo, 2024-09-27:
Could someone share binaries with this fix? https://github.com/wiredopposite/tusb_gamepad/pull/1
I can't get this project to compile on my computer.

baabo, 2024-10-04:
I was able to get the project to compile, and here are the binaries with the fix until it is hopefully included in a new release. This fix fixes the issue on my end as well.
[OGX-Mini v0.4.0 bchid fix.zip](https://github.com/user-attachments/files/17264030/OGX-Mini.v0.4.0.bchid.fix.zip)

Invictaz, 2024-11-10:
> I noticed with my 8bitdo v2 that I had voltage drops from the rp2040. I ran a wire to bypass to the usb host just to make sure that the 8bitdo had steady voltage. Running through the rp2040 it would drop from 5 to 3.7 volts. This has improved reliability in connection.

You have a photo of that?

IzWhatitizz, 2024-11-10:
> > I noticed with my 8bitdo v2 that I had voltage drops from the rp2040. I ran a wire to bypass to the usb host just to make sure that the 8bitdo had steady voltage. Running through the rp2040 it would drop from 5 to 3.7 volts. This has improved reliability in connection.
> 
> You have a photo of that? @IzWhatitizz

[image]
 It was not easy to do on a usb c connector, but works great. Another option is a breakout board of sorts. I also have another one I made with a cheap usb hub. One of those ones about the size of a flash drive.

Invictaz, 2024-11-10:
> > > I noticed with my 8bitdo v2 that I had voltage drops from the rp2040. I ran a wire to bypass to the usb host just to make sure that the 8bitdo had steady voltage. Running through the rp2040 it would drop from 5 to 3.7 volts. This has improved reliability in connection.
> > 
> > 
> > You have a photo of that? @IzWhatitizz
> 
> ![20241109_221226](https://private-user-images.githubusercontent.com/163659645/384657886-db769a13-b807-48b5-97ea-072e7823dc63.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzEyNTU3MjcsIm5iZiI6MTczMTI1NTQyNywicGF0aCI6Ii8xNjM2NTk2NDUvMzg0NjU3ODg2LWRiNzY5YTEzLWI4MDctNDhiNS05N2VhLTA3MmU3ODIzZGM2My5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTEwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTExMFQxNjE3MDdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hMzA5NThkNTNmMWE4MmY0ZTg1YjI2NmVmMTJkM2ZiN2FkYjljY2NhOGRiMWY1MWY2ZWM3MTcxNjkyOTdiMDAxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.scAy4P0yjwrvJwhjBxlbwZh6uKmkfqku9WoTv4NwfCU) It was not easy to do on a usb c connector, but works great. Another option is a breakout board of sorts. I also have another one I made with a cheap usb hub. One of those ones about the size of a flash drive.

Where does the 5v power cable go to?

Invictaz, 2024-11-10:
I can confirm the 

> I was able to get the project to compile, and here are the binaries with the fix until it is hopefully included in a new release. This fix fixes the issue on my end as well. [OGX-Mini v0.4.0 bchid fix.zip](https://github.com/user-attachments/files/17264030/OGX-Mini.v0.4.0.bchid.fix.zip)

I can confirm this build works, thanks for compiling.

IzWhatitizz, 2024-11-10:
@Invictaz  so 5v goes to the usb  female that your controller or Bluetooth adapter plugs into. Typical the four wires from the pi zero would go to the usb female. After testing output voltage from the pi in use, I  could see the 5v to 3.7v drop. This isn't an issue with wired controllers, and isn't a major issue with Bluetooth adapters, but by bypassing the 5v to the female usb has improved stable power to the Bluetooth adapter. I know 3.7v should be enough power, but the adapter for some reason doesn't like the random drops.

Invictaz, 2024-11-13:
> @Invictaz so 5v goes to the usb female that your controller or Bluetooth adapter plugs into. Typical the four wires from the pi zero would go to the usb female. After testing output voltage from the [...]

Yes but where do you connect the power lead from the female port to, to a male somewhere else?

IzWhatitizz, 2024-11-13:
@Invictaz I soldered it to the 5v pad on the pico zero's connector port. Another option is something like this. I use it for testing mostly. 
[image]

purplebar0, 2024-12-21:
Unfortunately, I wasn't so lucky. Neither a DualShock 3, nor an Xbox 360-compatible controller were usable on my unit which has UsbdSecPatch set as a launch plugin in DashLaunch.

Since said DS3 worked just fine when converted w/ OGX-Mini to an Xbox 360 controller when trying it on my computer, the plugin is definitely the culprit. I even tried the patched version mentioned earlier.

I know this isn't directly related to this project, but do you think it'd work if I patched XeBuild with it instead? Other plugins have always worked when I added them to DashLaunch list, so I don't see why this wouldn't.

lopenator, 2024-12-21:
> Unfortunately, I wasn't so lucky. Neither a DualShock 3, nor an Xbox 360-compatible controller were usable on my unit which has UsbdSecPatch set as a launch plugin in DashLaunch.
>
> Since said DS3 [...]

works for me. how bizarre

purplebar0, 2024-12-21:
Update: even with XeBuild that includes UsbdSec patch, I couldn't get this to work. At least if this does get working, XeBuild approach would load the patch immediately at boot, so I take it as a bonus.

Invictaz, 2025-01-01:
Problems should be fixed now in Alpha 2

purplebar0, 2025-01-01:
> Problems should be fixed now in Alpha 2

Can confirm. 1.0 Alpha 1 had an issue that made connecting impossible, albeit nobody brought it up in this issue.