wiredopposite/OGX-Mini #24: v0.4.0

Androxilogin, 2024-04-25:
I just had one heck of a time with this. It seems the 2040-Zero file doesn't work on the Zero, the regular one does.

wiredopposite, 2024-04-25:
Hey the usb pins are different for the rp2040-zero file, data +/- are GPIO 10 and 11, did you have the port hooked up to those?

Androxilogin, 2024-04-26:
Actually, no! I was unaware of that. It worked with the regular pins with the other file on the RP2040-Zero. Is there a consequence in using it this way? I'll switch them over now, regardless but it seems to work this way.

wiredopposite, 2024-04-26:
No there's no difference between either firmware except the pins, I just changed them for the RP2040-Zero so you can attach the USB A port to the end opposite the USB C port. Also for the custom PCB in the [hardware folder](https://github.com/wiredopposite/OGX-Mini/tree/master/hardware).

Androxilogin, 2024-04-27:
Ah, I see. Working as intended. Awesome project!

closed 2024-08-17 by wiredopposite