wiredopposite/OGX-Mini #25: Add Logitech RumBlepad 2 USB wired

danko79, 2024-04-25:
Hi, i have a Logitech RumblePad 2 USB wired, but not working it's possible to add?
[image]

    =========================== USB Port1 ===========================

Connection Status        : 0x01 (Device is connected)
Port Chain               : 4-1

      ========================== Summary =========================
Vendor ID                : 0x046D (Logitech Inc.)
Product ID               : 0xC218
USB Version              : 1.1
Port maximum Speed       : Full-Speed
Device maximum Speed     : Low-Speed
Device Connection Speed  : Low-Speed
Self powered             : no
Demanded Current         : 500 mA
Used Endpoints           : 3

      ======================== USB Device ========================

        +++++++++++++++++ Device Information ++++++++++++++++++
Device Description       : Dispositivo di input USB
Device Path              : \\?\USB#VID_046D&PID_C218#5&28a8b183&0&1#{a5dcbf10-6530-11d2-901f-00c04fb951ed} (GUID_DEVINTERFACE_USB_DEVICE)
Kernel Name              : \Device\USBPDO-10
Device ID                : USB\VID_046D&PID_C218\5&28A8B183&0&1
Hardware IDs             : USB\VID_046D&PID_C218&REV_0100 USB\VID_046D&PID_C218
Driver KeyName           : {745a17a0-74d3-11d0-b6fe-00a0c90f57da}\0031 (GUID_DEVCLASS_HIDCLASS)
Driver                   : \SystemRoot\System32\drivers\hidusb.sys (Version: 10.0.19041.3636  Date: 2023-10-29)
Driver Inf               : C:\WINDOWS\inf\input.inf
Legacy BusType           : PNPBus
Class                    : HIDClass
Class GUID               : {745a17a0-74d3-11d0-b6fe-00a0c90f57da} (GUID_DEVCLASS_HIDCLASS)
Service                  : HidUsb
Enumerator               : USB
Location Info            : Port_#0001.Hub_#0006
Location IDs             : PCIROOT(0)#PCI(1600)#USBROOT(0)#USB(1), ACPI(_SB_)#ACPI(PCI0)#ACPI(USB5)#USBROOT(0)#USB(1)
Container ID             : {5fe6036e-032e-11ef-b74e-a42b8ce8bd51}
Manufacturer Info        : (Periferiche standard di sistema)
Capabilities             : 0x84 (Removable, SurpriseRemovalOK)
Status                   : 0x0180600A (DN_DRIVER_LOADED, DN_STARTED, DN_DISABLEABLE, DN_REMOVABLE, DN_NT_ENUMERATOR, DN_NT_DRIVER)
Problem Code             : 0
HcDisableSelectiveSuspend: 0
EnableSelectiveSuspend   : 0
SelectiveSuspendEnabled  : 0
EnhancedPowerMgmtEnabled : 1
IdleInWorkingState       : 0
WakeFromSleepState       : 0
Power State              : D0 (supported: D0, D2, D3, wake from D0, wake from D2)
 Child Device 1          : Controller gioco compatibile HID
  Device Path            : \\?\HID#VID_046D&PID_C218#6&36f24dae&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030} (GUID_DEVINTERFACE_HID)
  Kernel Name            : \Device\000000a8
  Device ID              : HID\VID_046D&PID_C218\6&36F24DAE&0&0000
  Class                  : HIDClass
  Driver KeyName         : {745a17a0-74d3-11d0-b6fe-00a0c90f57da}\0032 (GUID_DEVCLASS_HIDCLASS)

        ---------------- Connection Information ---------------
Connection Index         : 0x01 (Port 1)
Connection Status        : 0x01 (DeviceConnected)
Current Config Value     : 0x01 (Configuration 1)
Device Address           : 0x03 (3)
Is Hub                   : 0x00 (no)
Device Bus Speed         : 0x00 (Low-Speed)
Number Of Open Pipes     : 0x02 (2 pipes to data endpoints)
Pipe[0]                  : EndpointID=1  Direction=IN   ScheduleOffset=5  Type=Interrupt  wMaxPacketSize=0x8     bInterval=10  -> 1568 Bits/ms = 196000 Bytes/s
Pipe[1]                  : EndpointID=1  Direction=OUT  ScheduleOffset=6  Type=Interrupt  wMaxPacketSize=0x8     bInterval=10  -> 1568 Bits/ms = 196000 Bytes/s
Data (HexDump)           : 01 00 00 00 12 01 10 01 00 00 00 08 6D 04 18 C2   ............m...
                           00 01 01 02 00 01 01 00 00 03 00 02 00 00 00 01   ................
                           00 00 00 07 05 81 03 08 00 0A 05 00 00 00 07 05   ................
                           01 03 08 00 0A 06 00 00 00                        .........
[... 91 lines cut ...]
bmAttributes             : 0x03 (TransferType=Interrupt)
wMaxPacketSize           : 0x0008
bInterval                : 0x0A (10 ms)
Data (HexDump)           : 07 05 01 03 08 00 0A                              .......

      -------------------- String Descriptors -------------------
             ------ String Descriptor 0 ------
bLength                  : 0x04 (4 bytes)
bDescriptorType          : 0x03 (String Descriptor)
Language ID[0]           : 0x0409 (English - United States)
Data (HexDump)           : 04 03 09 04                                       ....
             ------ String Descriptor 1 ------
bLength                  : 0x12 (18 bytes)
bDescriptorType          : 0x03 (String Descriptor)
Language 0x0409          : "Logitech"
Data (HexDump)           : 12 03 4C 00 6F 00 67 00 69 00 74 00 65 00 63 00   ..L.o.g.i.t.e.c.
                           68 00                                             h.
             ------ String Descriptor 2 ------
bLength                  : 0x32 (50 bytes)
bDescriptorType          : 0x03 (String Descriptor)
Language 0x0409          : "Logitech RumblePad 2 USB"
Data (HexDump)           : 32 03 4C 00 6F 00 67 00 69 00 74 00 65 00 63 00   2.L.o.g.i.t.e.c.
                           68 00 20 00 52 00 75 00 6D 00 62 00 6C 00 65 00   h. .R.u.m.b.l.e.
                           50 00 61 00 64 00 20 00 32 00 20 00 55 00 53 00   P.a.d. .2. .U.S.
                           42 00                                             B.

Invictaz, 2024-04-26:
Maybe this also paves the way for the famous F310? It looks almost the same.

danko79, 2024-04-26:
The F310 is the new model.

wiredopposite, 2024-05-02:
[ogx-mini_rumblepad.zip](https://github.com/wiredopposite/OGX-Mini/files/15192491/ogx-mini_rumblepad.zip)
Can you see if this works for your? I don't have this controller to test

danko79, 2024-05-02:
> [ogx-mini_rumblepad.zip](https://github.com/wiredopposite/OGX-Mini/files/15192491/ogx-mini_rumblepad.zip) Can you see if this works for you? I don't have this controller to test

Try this on rpi pico not working, only the vibration button works, if press it the controller vibrates.
the other buttons do not respond to commands.

closed 2026-09-19 by danko79