wiredopposite/OGX-Mini #27: Wrong mapping in D-Input mode...

CloudyShane, 2024-05-03:
**Describe the bug**
When I connect my gamepad to OGX Mini, it boots in D-Input mode. Most of the buttons are mapped correctly, but the most important ones are swapped:

Controller press --- What Xbox sees:

A Button --- B Button
B Button --- A Button
X Button --- Y Button
Y Button --- X Button

(I'm using Xbox Controller Tool to see what is what)

**What type of controller do you have the issue with?**
I'm using my trusted MPOW Wired PS4 (Supports X-Input mode) controller from the previous issue. VID and PID are 0x2563 and 0x0357 when in PS4 mode. Yet it started working in D-Input mode on version 0.4.0.

**What platform are you using the OGX-Mini on?**
Original Xbox

**What board are you using?**
RP 2040 Zero from Aliexpress. The 0.4.0 release for the RP 2040 Zero didn't work, but the release 0.4.0 for Pico does! I'm using that version right now.

Invictaz, 2024-05-03:
Maybe the mapping got copied from the Switch output?

CloudyShane, 2024-05-03:
It might have been, but I'm clueless. Anyway, the PS4 mode actually works on OGX Mini and has all the correct button mappings. So, only D-Input is in question.

wiredopposite, 2024-05-03:
Is the OGX-Mini in PS3 mode? Because the mappings in that mode show up that way on PC but are correct on PS3, same with Switch mode

Invictaz, 2024-05-03:
So it must be in ps3 mode to have correct output? And what if you want to translate dinput to xinput like @CloudyShane wants to do?

wiredopposite, 2024-05-03:
> So it must be in ps3 mode to have correct output? And what if you want to translate dinput to xinput like @CloudyShane wants to do?

I don't have the controller so I can't really say, all the DInput controllers I've tested have a mapping quirk with the action buttons so I'd have to manually change those for certain ones without the issue. I am working on a way to remap buttons so the user can just take care of the issue on their own though. 

In the meantime I'll edit the mappings and post firmware in this thread when I get the chance later on today, that way we'll see if the issue is what I believe it be.

CloudyShane, 2024-05-03:
> Is the OGX-Mini in PS3 mode? Because the mappings in that mode show up that way on PC but are correct on PS3, same with Switch mode. Also I should update the readme to be clearer, the pinout of the rp2040-zero is different, gpio 10/11 are d +/-

No, the gamepad was in Original Xbox mode. No matter the mode, the buttons are swapped. I tried X-Input, Switch...

> all the DInput controllers I've tested have a mapping quirk with the action buttons so I'd have to manually change those for certain ones without the issue

Sadly, I can't really determine the VID and PID combo for my gamepad while in DInput mode because I can't find a way to turn said mode for use on PC.

wiredopposite, 2024-12-22:
Mappings can now be changed with the webapp

closed 2024-12-22 by wiredopposite