wiredopposite/OGX-Mini #28: Switch OG Xbox white and black button mapping

DenshaMan, 2024-05-03:
Is there a way to swap how the original xbox black and white buttons are mapped to LB and RB on modern Xbox controllers? Having black and white on LB and RB in general is already a godsend, but in some games having the option to have the mapping reversed would be wonderful.

Thanks for this awesome project!

wiredopposite, 2024-05-06:
Not currently, but I'm working on a way to plug the adapter into a PC and choose different button mappings using a webpage. I have the firmware side basically done and I'm just getting started on the webapp, it'll be ready fairly soon.

DenshaMan, 2024-05-06:
Thanks for the quick reply - that sounds really awesome! With the way your tool is progressing, I wonder if it could become a full alternative to something like the Titan Two or Chronus Max at some point. (You're already quite a ways there!) Since at least the Titan Two is not being actively supported anymore, there's definitely a need for a USB controller translator. If you accept the submission of USB HID descriptors, I have quite a few obscure controllers that I'd *love* to see supported with output/translation to Nintendo Switch and PS4.

closed 2024-12-22 by wiredopposite