wiredopposite/OGX-Mini #30: don´t dettect ps mini controller

EvlShdw, 2024-05-14:
following the provided diagram https://github.com/wiredopposite/OGX-Mini/blob/master/images/pi_pico_diagram.png the pi pico don’t recognize the ps mini controller but the pico is recognized as xbox og controller even in pc

EvlShdw, 2024-05-14:
using version 0.4.0

Invictaz, 2024-05-14:
Controller model number is SCPH-1000R

EvlShdw, 2024-05-15:
that's the model of my controller

EvlShdw, 2024-05-15:
> Controller model number is SCPH-1000R

that´s my model number

EvlShdw, 2024-05-19:
allready test with xbox one controller and did´t work

wiredopposite, 2024-05-21:
Which firmware file are you using, the one labeled Pico? I don't have any reproduction boards from aliexpress so I'm not positive if there would be issues with those

EvlShdw, 2024-05-21:
firmware 0.4.0 OGX-Mini_Pico.uf2

wiredopposite, 2024-12-22:
You every figure this out? Wrong usb pins maybe?

closed 2025-01-07 by wiredopposite