wiredopposite/OGX-Mini #32: Xbox 360 Receiver, Third Party, Connected once on Xbox OG - stuck syncing

CrabBotherer, 2024-05-26:
**Describe the bug**
Hi team, I have a old 3rd party 'Chinese' Xbox 360 Receiver - USB based. Hardware PID's are;

USB\VID_045E&PID_0291&REV_0107
USB\VID_045E&PID_0291

I am using a Raspberry Pi Pico-W, have double checked my wiring of USB, and have tested it with other controllers fine (Wired PS4, Wired Xbox One) working great.

However, using the wireless receiver, I got it to connect once using the Xbox OG, but after powering on Xbox via a accidental eject button power on, it stopped working.

I reflashed the firmware I compiled (attached - rename file to .uf2) but the same issue occurred.

A note - when I plugged the Pico-W into my laptop via USB, after flashing the firmware, the controller would sync fine. Odd!

Happy to provide any files/logs (if any) to assist. Sincere thanks for this project!

[OGX-Mini.txt](https://github.com/wiredopposite/OGX-Mini/files/15446163/OGX-Mini.txt)

**What type of controller do you have the issue with?**

Xbox 360 Wireless Controllers - three different ones, same issue occurred on all.
 

**What platform are you using the OGX-Mini on?**

 OG Xbox

**What board are you using?**

Pi Pico-W

CrabBotherer, 2024-06-01:
A quick update from my side - I went out and got a actual Pico - without the Wireless/Bluetooth - flashed the preconfigured .uf2 from your releases tab to it and had the same issue. 

Looking over my wiring, I realized I was 'cheating' with the Ground, wiring it to 38 instead of your 23. Thinking 'Well, you should follow the instructions Crabs!' I rewired it and had the same issue.

But- when I connected my Xbox 360 3rd Party USB *after* boot on the Xbox - I got the LED lighting up and it worked fine! So, interesting! I'll keep mucking around but thought I'd provide a quick update. Thanks again for this neat project!

wiredopposite, 2024-12-22:
There have been times when I've had to reseed the controller for it to work after powering on the ogxmini, does they new firmware do the same thing?

closed 2025-01-07 by wiredopposite