wiredopposite/OGX-Mini #33: Add wireless controller Retrobit Sega Saturn Controller for OGxbox

captnkarma, 2024-06-04:
I am frustrated the retrobit sega saturn controller is not usable. It´s recognized and pairable but no control.
Also 8bitdo M30 didn´t work wirelessly with its own dedicated usb stick no blutooth).

Could you add the controller ID for the Retrobit Seaga Saturn controller as it has a USB stick meant for windows OR explain how to add it?

How did you add additional controllers anyways? Maybe I can provide some time and work. Xbox one controller works flawless
but its the only one at 100% compatibility. Tried only wired yet. A Ps2 usb and a N64 usb controller did work. Snes usb didn´t. 8bitdo M30 did only recognize buttons but not the d-pad. Ordered a 8bitdo wireless-2 usb to further investigate wireless. Thanks for the hard work!

wiredopposite, 2024-08-15:
If you have Windows, you can plug it in, open Device Manager, then look under Human Interface Devices. It should be in there, it'll probably say "HID-compliant game controller", right click that and select Properties, go to the Details tab, then select Hardware IDs from the dropdown menu, it should have VID and PID listed there so post that if you can. 

Also try opening Steam, go to Settings > Controller and see what kind of controller it's showing up as, it should be next to "Controller Name", that would be helpful since I don't own one of these and I'd have to know what kind of driver it'd be using before adding support for it.

captnkarma, 2024-11-19:
HID\VID_0079&PID_0006&REV_0107
HID\VID_0079&PID_0006
HID\VID_0079&UP:0001_U:0004
HID_DEVICE_SYSTEM_GAME
HID_DEVICE_UP:0001_U:0004
HID_DEVICE

captnkarma, 2024-11-19:
It just says "USB Joystick" in Steam.

captnkarma, 2024-11-19:
Sorry for getting back at you this late ... actually I bought some 3rd party xbox controller for PC on amzn and they worked flawlessly BUT I was only testing and still had to repair the console, so I wasn´t actually playing much. But when I played THPS 2x I recognized quite a bit of latency. So momentarily I am in the process of finding out what is causing the latency... the controller or maybe some wrong settings in the console. Anyways your code works enhancement of the code works perfectly and I hope I can fix the latency issues and then print some cases finally and play :D Thanks for your hard work you´re the man!

captnkarma, 2024-11-19:
PS: Actually the 8bit do 2 usb stick works together with a xbox 360 one wireless which had faster input/less latency so did some testing and found that probably the 3rd party controller´s latency source is the mechanical part of the gamepad make the way for the switches too long and so introducing latency... I´ll try a mod on it to make it better.
PPS: Regarding the sega saturn controller of opening the issue in the first place: I just figured its a complete different button layout that makes not much sense to play with on xbox nor playstation games (PS ported games like THPS 2x for example) Ok that´s all for now. Merry christmas already it came early this year! Thanks again for making it possible to play xbox again. I resurrected the console after 15 years but lost the controllers so even after I repaired the console there was no way to play without this project. All the best to you, Many wishes Cheers CK

captnkarma, 2025-01-07:
Hello wiredopposite, Happy new year! I modded that chinese xbox og controller of mazon with some thin cork matte both the d-pad and the buttons, now its basically on par with the xbox one controller. I then figured the analog pad was much more fast in reaction than the dpad. So I play THPS with that now. Problem solved. Thans again for your awesome work its such a joy to play with these wireless gamepads on xbox now and absolutely worth the work. Will mod the 2nd controller too. Message me if you want some foto material of my gamepad cork matte mod :-D Cheers