wiredopposite/OGX-Mini #36: Input mode - State - show with addressable led, library available already.

mrdude2478, 2024-06-16:
Add support that when you change the input mode state from for example XInput to Switch mode an addressable led such as ws2812b mapped to a gpio pin will change colour to reflect the mode you are in, this same led can be used to show contoller connection/disconnection state. A library already exists to do this whic can be found here:
https://github.com/blaz-r/pi_pico_neopixel

This should be fairly easy to implement looking at the library source and examples, so if you have time and cwant to add that feature it would be very much appreciated.

Additional request:
Most boards if not all nowadays can use addressable leds, also rasberry pi has 2 cores and this led code can be implemented to use the core not in use which wouldn't cause any issues with the main code that's runnning on the main core, this is of course probably not needed but is an option in case of running into any issues with controller input.

Invictaz, 2024-06-16:
This project has already Neopixel support maybe some code can be adapter

https://github.com/HandHeldLegend/GC-Adapter-RP2040

mrdude2478, 2024-06-17:
@Invictaz 

Thanks I'll have a look at that, also I see that it uses the ws2812 code that are in the piko-sdk examples.

Invictaz, 2024-06-22:
> @Invictaz
> 
> Thanks I'll have a look at that, also I see that it uses the ws2812 code that is in the piko-sdk examples.

Is it what you are looking for?
In my opinion all these Pico controller projects do some of the same work, namely controller translation. I hope some day someone will merge them all together to have one big good firmware instead of reinventing the wheel.