wiredopposite/OGX-Mini #39: Button Remapping, and Joystick Deadzone Change.

roxas10, 2024-08-14:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
I would love to change deadzone on joysticks because I use this adapter with PS3 and Xbox360 Wheels, and in some games I can't Drive Correctly cause has a giant deadzone.
I also need to remap some buttons so I can change the shift buttons.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Additional context**
Add any other context or screenshots about the feature request here.

wiredopposite, 2024-08-18:
A webapp for remapping and adjusting deadzones is coming in the next update. I've rewritten a lot of the firmware, which has taken a while, but it should be pretty soon, within the next week or two

mastercheff, 2024-08-19:
Also, some sort of Anti-DZ would be greatly appreciated , since most games default to ~25% , Monkey Ball DX is barely playable compared to gamecube port.

[image]

Invictaz, 2024-12-20:
Button remap is in the newest release v1.0 alpha

closed 2025-01-07 by wiredopposite

Invictaz, 2025-01-07:
Yes but deadzone and anti deadzone is missing so I'm not sure why this is closed?

mastercheff, 2025-01-07:
deadzone is implemented now We can add more DZ , reducing it would be most beneficial in most games. 2/3rds of the request are completed, and anti DZ is on the roadmap as I see it in the readme, We're cool

wiredopposite, 2025-01-07:
I'm not ignoring this, I've added it to the roadmap and want to keep the issues page from getting cluttered with too many things that aren't actual problems with the firmware, it makes it harder to parse through. Also the original issue wasn't about anti-dz.