wiredopposite/OGX-Mini #41: Powera xbox controller does not connect or turn on

gaberilde, 2024-08-29:
**Describe the bug**
Powera xbox one controller does not connect or even turn on the light and does not work
vid pid and some other info below let me know if you need more.

**What type of controller do you have the issue with?**
  idVendor            0x20d6
  idProduct           0x2016
  bcdDevice           0101
  iManufacturer       1     PowerA
  iProduct            2     Xbox Series X EnWired Green Hint

**What platform are you using the OGX-Mini on?**
I tried Switch and pc

**What board are you using?**
Adafruit Feather, RP2040 usb

Other controllers like ps3 worked

wiredopposite, 2024-09-03:
I'd think this should work assuming it works with a real Xbox, I don't own one of these though. To know for sure, could you install [Wireshark](https://www.wireshark.org/download.html)? Assuming you're on Windows, you'd need to open that, double click USBPcap1 and capture some USB traffic with the gamepad plugged in, a very short time is fine and you can click the stop button. If you could do that and then File > Save As and upload that file, it would be helpful. If it doesn't let you upload it here, you probably just need to save it to a zip file and upload that.

gaberilde, 2024-09-03:
> I'd think this should work assuming it works with a real Xbox, I don't own one of these though. To know for sure, could you install [Wireshark](https://www.wireshark.org/download.html)? Assuming [...]

hi i cant seem to find that, do you know where it would be?

Invictaz, 2024-09-03:
https://www.wireshark.org/download.html

gaberilde, 2024-09-03:
> https://www.wireshark.org/download.html

No i have wire shark but can't find usbpcap1

wiredopposite, 2024-09-03:
> > https://www.wireshark.org/download.html
> 
> No i have wire shark but can't find usbpcap1

I think there was an option in the Wireshark installer to install USBPcap along with it, but it's been a while, you might just need to manual install it https://desowin.org/usbpcap/

gaberilde, 2024-09-03:
> > > https://www.wireshark.org/download.html
> > 
> > 
> > No i have wire shark but can't find usbpcap1
> 
> I think there was an option in the Wireshark installer to install USBPcap along with it, but it's been a while, you might just need to manual install it https://desowin.org/usbpcap/
> 
> After that it should show up in Wireshark when you open it

[image]

Nope no usb pcap option on mine after reinstalling and rebooting also