wiredopposite/OGX-Mini #43: "Hyperkin Duchess" controller shows light, but does not work.

BlazeModz, 2024-10-05:
**Describe the bug**
I bought the Modvilleusa "ogxmini" adapter and it works amazingly! I plug in my 360 controller and it works just great! However I decided to buy a Hyperkin Duchess to use exclusively with my OG Xbox on this adapter and, well, the light shows up when I plug it in, but I can't actually use the controller, none of my inputs are registered. I tried on XInput mode on the adapter, as well as OG Xbox mode.

**What type of controller do you have the issue with?**
Hyperkin Duchess, here's what's listed in Device Manager for the VID and PID: VID_2E24&PID_0423

**What platform are you using the OGX-Mini on?**
 Original Xbox!

**What board are you using?**
Unsure, I bought it from here https://modzvilleusa.com/products/xbox-plug-and-play-ogx-mini-ogx360 but from what I can tell it seems to be RP2040?

wiredopposite, 2024-12-15:
I don't have this controller so I'd need a USB descriptor dump from someone to see if it's different from a normal Xbox one/series controller. If you're on Windows you can install Wireshark, while you're installing it make sure you allow it to install USBPcap as well. 

Open up Wireshark, select USBPcap, you should see it in the Capture menu right when you open the program. It'll start capturing USB traffic, then plug in your Duchess controller, it'll show a bunch of lines that start with "GET DESCRIPTOR", then you can press the red square at the top of the window to stop the capture. Go to File > Save and save the file, upload it here and I'll take a look at it. 

I don't have one of the Modzville adapters so I can't say if it has a button to load new firmware, you'd have to look for one on there or ask him about it.

baronunread, 2024-12-16:
I don't have the Modzville adapart either, I built my own OGX-Mini by hand so I can't help on that regard
I'll install Wireshark later and provide the descriptor dump! @wiredopposite

baronunread, 2024-12-16:
@wiredopposite For some reason pcapng files aren't supported by github! I compressed it into a Zip file
[Duchess.zip](https://github.com/user-attachments/files/18155252/Duchess.zip)

wiredopposite, 2024-12-16:
> @wiredopposite For some reason pcapng files aren't supported by github! I compressed it into a Zip file [Duchess.zip](https://github.com/user-attachments/files/18155252/Duchess.zip)

Thanks, it does just look like an Xbox One controller. I've had issues with some controllers like Switch Pro where the USB cable is inadequate for powering it, so also make sure that USB cable is okay (works without the OGX-Mini) and the wire between the USB-A port and+5V from the board is thick enough, 22awg is what I'll usually use. I might just have to buy one of these to figure it out though, the power thing might have nothing to do with this issue.

Tell me what uf2 file you're using for your board and I'll post the new firmware here for your to try, I did rewrite most of the XInput driver so it might behave differently now.

baronunread, 2024-12-16:
> Tell me what uf2 file you're using for your board and I'll post the new firmware here for your to try, I did rewrite most of the XInput driver so it might behave differently now.

OGX-Mini_RP2040-Zero.uf2

mastercheff, 2024-12-16:
I feel like I need to chime in.

I had similar problem with Gamesir Nova Lite controller, turned out my pico (generic usbC 4MB board) was not outputting steady 5 Volts on VBUS (40) pin, it was closer to 4,5 V under load. Gamepad behaves normally in windows (XInput 360 controller) somehow only when I resoldered corresponding wire to USB host,  controller started to work nominally

wiredopposite, 2025-01-22:
I just ordered a Duchess so I should have it working soon

baronunread, 2025-01-23:
> I just ordered a Duchess so I should have it working soon

Oof I forgot to follow up! I ended up making it work...
I sent you some messages on discord but you weren't answering and I kind of forgot that I had opened the issue, I'm sorry!

I built a small PCB that grabs the 5V directly from the console's traces and brings it to the controller USB A port and that made it work, I was able to use the duchess with no issues whatsover.

I feel kinda bad though do you have a patreon? I'll send some money over for your dedication ❤ @wiredopposite

xevman, 2025-01-26:
I’m having the same issue with my Duchess, looking forward for any potential future fixes.

mastercheff, 2025-01-26:
the fix is to feed it stable 5 volts. Some pico boards are dodgy (their voltage regulator)

mastercheff, 2025-01-26:
USB 5v line where You connect the Controller to, should go to places marked instead of pin 40