wiredopposite/OGX-Mini #44: Any chance of support Google Stadia Controller?

impeeza, 2024-10-08:
The google stadia even with the released firmware uses a different PID and probably protocol,  there is any chance will be supported?

[image]

[image]

The controller is detected by the Joy.cpl applet on windows, but not by gamepad tester (https://hardwaretester.com/gamepad)
[image]

let me know if I can help on any further action.

Thanks.

mrdude2478, 2024-11-01:
I use this as an OG xbox controller (which is default), you need these drivers installed and this program for it to show up button presses on a PC:

https://www.s-config.com/xbcd-original-xbox-controllers-win10/

wiredopposite, 2025-01-22:
> The google stadia even with the released firmware uses a different PID and probably protocol, there is any chance will be supported?
>
> [image]
>
> [image]
>
> The controller is detected by the Joy.cpl [...]

Can you try out alpha 3 and see if it works? I added generic HID support so it's possible it'll work with a stadia controller

Kanadeforever, 2025-09-22:
Version 1.0.0-alpha3 supports the Stadia controller and also properly supports vibration functionality in Bluetooth mode. However, the Google Assistant button and the screenshot button are not functional. Is it possible to allow customization of these two buttons in the web app in future versions? Or perhaps enable them to be mapped to keyboard keys?