wiredopposite/OGX-Mini #45: Mount LED not working with RP2040 Zero + Interposer Board

SRDWA, 2024-10-09:
Hello,
as you can read in the title, the mount/unmount LED is not working.

I have already found the error. See screenshot.
The LED on the interposer board uses GPIO 14 and not GPIO 13.
Unfortunately I cannot compile the firmware myself.
I would like a correction.

Kind regards, Sven

[image]

danko79, 2024-10-11:
Yes, i have a same problem.

Dean24769, 2024-10-18:
Use a jump wire from 5v on the chip to pin 14 to bypass

SRDWA, 2024-10-19:
@Dean
But your solution is not correct.
1. You would permanently destroy the GPIO 14 output in the microcontroller!
2. The LED would be permanently lit.

My temporary solution is:
Cut the trace between GPIO 13 and the resistor R3.
I used a 1mm drill bit for this.
Then connect R3 to GPIO 13 with a jumper wire.

Pictures follow.

SRDWA, 2024-10-19:
[image]
[image]

danko79, 2024-10-19:
ok for the solution, but I didn't understand if the problem is in the pcb or in the firmware?

SRDWA, 2024-10-19:
Which came first, the chicken or the egg?

I think the firmware was there first. So the error is on the interposer PCB.
I can only see 2 solutions to fix the problem.
1. Either the interposer PCB is changed and the LED is connected to GPIO 13, or
2. An agreement is reached and GPIO 14 is used in future firmwares.

As there seems to be no official support here, I have chosen the hardware solution. I am unable to fix the error on the software level.

danko79, 2024-10-19:
> Which came first, the chicken or the egg?
>
> I think the firmware was there first. So the error is on the interposer PCB. I can only see 2 solutions to fix the problem.
>
>     1. Either the interposer [...]

Let's hope you update the firmware then I don't want to modify the pcb.
Thanks

AR1972, 2024-11-28:
I fixed this
https://github.com/AR1972/OGX-Mini/releases

danko79, 2024-11-28:
> I fixed this https://github.com/AR1972/OGX-Mini/releases

Works fine! Many thanks.

wiredopposite, 2024-12-22:
The new firmware sets the correct pin now, and also the rgb led on there

closed 2024-12-22 by wiredopposite