wiredopposite/OGX-Mini #46: 8bitdo Ultimate C Wired not working

mastercheff, 2024-10-14:
**Describe the bug**
8bitdo Ultimate C Wired controller is not working for me,  newer Ultimate 2C is working great though . Home button diode blinks so it's at least getting power

**What type of controller do you have the issue with?**
8bitdo Ultimate C Wired

Device Manager throws this as HW ids :

USB\VID_2DC8&PID_3106&REV_0114
USB\VID_2DC8&PID_3106

**What platform are you using the OGX-Mini on?**
 OG Xbox, PC

**What board are you using?**
RP2040-Zero , Unit behaved well with both Xbox and PC in XInput mode

Invictaz, 2024-12-20:
Would you be able to do a USB descriptor dump and post it here?

mastercheff, 2024-12-21:
the pad is at my Buddys rn, will get to it eventually, rather soon

mastercheff, 2024-12-26:
closed, looks duplicate with #58

closed 2024-12-26 by mastercheff