wiredopposite/OGX-Mini #51: BUG Keep pairing after mode change

Invictaz, 2024-12-20:
For now after each mode change (e.g. Xbox output to PS3 output) the controller tries to connect and the Pico goes back to flashing led mode. You have to manually turn off the controller, then do re-pairing. 

The pairing should stay, only the output should change. 
This works on 8bitdo Adapter Pro 2 so should be possible.

wiredopposite, 2024-12-20:
Tinyusb has to be deinitialized in order to safely write new settings to the flash and doesn't seem to support being reinitialized without hard a hard reboot. I can work on getting bluetooth safely disconnected before rebooting though, so it doesn't get stuck on.

Invictaz, 2024-12-20:
> Tinyusb has to be deinitialized in order to safely write new settings to the flash and doesn't seem to support being reinitialized without hard a hard reboot. I can work on getting bluetooth safely disconnected before rebooting though, so it doesn't get stuck on.

I understand but why does the 8bitdo device work then? I mean the bluetooth mac address can be stored in flash memory so it will connect to the last used controller?

mastercheff, 2024-12-20:
It's not critical, but I noticed that too a while ago. Tomorrow I will be testing out controllers that didn't work for me (8bitdo Ultimate C Wired) We Lannin' Halo CE Baby