wiredopposite/OGX-Mini #56: BUG OGX-Mini-v1.0.0a-PI_PICOW.uf2 cannot use USB port

Invictaz, 2024-12-20:
If you flash Pico W uf2 to pico W you have bluetooth
If you flash Pico uf2 to pico W you have USB

Pico W (and Pico 2 W) should have USB access. Simply put if you disconnect the usb cable it goes back into pairing mode and you can use the wireless option, if you turn off a wireless controller and plug in a wired one it will continue wired.

wiredopposite, 2024-12-21:
That's not a bug, the usb host stack isn't being used in the pico w firmware, maybe in a future version

Invictaz, 2024-12-21:
I hope so.

purplebar0, 2024-12-21:
Other than supporting both wireless and wired gamepads, implementing USB host would pave the way for DS3 pairing, if it is planned for the future. Manual pairing would otherwise [require connecting the DS3 to a computer and running `sixaxispairer`](https://bluepad32.readthedocs.io/en/latest/pair_ds3/) whenever the board wasn't the last paired device.

Invictaz, 2024-12-23:
@purplebar0 I agree. See #17

Lgmrszd, 2025-08-02:
Could this be implemented as a switch in the web interface? This way, it would be easy to keep functionality of both usb and wireless connection without possible unpredictable behavior when attempting to use both wired and wireless connection.

Lgmrszd, 2025-08-06:
> > Could this be implemented as a switch in the web interface? This way, it would be easy to keep functionality of both usb and wireless connection without possible unpredictable behavior when attempting to use both wired and wireless connection.
> 
> It can be implemented pretty easily with the new alpha I've been writing so I can include an option to turn it on/off. USB host and device must run on the same core since bluetooth needs its own, that'll cause some more latency if it's enabled, if it's disabled it wouldn't add any though

Thanks for the update! Looking forward to this