wiredopposite/OGX-Mini #57: Pi-Pico to ESP32 wiring?

mrdude2478, 2024-12-21:
Would it be possible to add a picture with wiring from Pi-Pico to ESP32?

Is this how it should be wired?

GPIO pico > GPIO esp32
#define I2C_SDA_PIN     18 (pin 24) // esp32 SCL
#define UART0_TX_PIN    16 (pin 21) // esp32 RX
#define UART0_RX_PIN    17 (pin 22) // esp32 TX
#define MODE_SEL_PIN    21 (pin 27) // Not Connected to esp32
#define ESP_PROG_PIN    20 (pin 26) // ESP32 IO0
#define ESP_RST_PIN     8  (pin 11) // ESP32 EN

I am unsure as I thought you would need both SCL & SDA on the esp32 to be wired to SDA/SCL on the Pico. Probably I am reading the code wrong. A picture would help make this clearer. Thanks for adding bluetooth support to this, it's great.

Also I have a question regarding BT, would it be possible to add a chip that has classic bluetooth and BLE so that you can use pretty much any controller? I am very happy with classic bluetooth as all my controllers support this.

wiredopposite, 2024-12-21:
Yes I'm gonna be posting a diagram with the wiring 
when i get back to my computer after the holidays, along with binaries. The pico's scl is connected, it's sda+1 so 19, booting the pico with  mode_sel low tells it to put the esp32 into programming mode and makes the pico act as a uart bridge so you can program the esp32 through it

mrdude2478, 2024-12-22:
Thanks for replying and confirming the pinouts. That's good you will post the bins. I managed to get the esp32 bin compiled in windows 10 with Platformio (which took me a while (a few hours) to figure out as it wasn't downloading BT stackautomatically and there's different versions of that). For the PICO I couldn't figure out how to compile with platformio but I managed OK using ubuntu in a VM.

I have another question if you're not too busy, I noticed that when compiling for ESP32 the USB input with a dongle on the PICO no longer works, it would be good if esp32 bluetooth wasn't connected if could send a signal to use the USB as before as I use a 5GHZ usb dongle with EasySMX X20 controller which works great, and also a Mayflash Magic-S Ultimate (which works on PS4), It would be good to have both USB and Bluetooth support if this is possible. Also I see you added addressable RGB led support (thanks), I didn't check the code much but can you have that so that depending on which mode you are in (switch/ps3/xbox etc), the addressable led changes colour as a visual aid to let you know the mode?

Invictaz, 2025-01-01:
@mrdude2478 seems included now in alpha 2

closed 2025-01-07 by wiredopposite