wiredopposite/OGX-Mini #58: 8BitDo Ultimate C 2.4GHz (Wireless) not working

faithvoid, 2024-12-21:
**Describe the bug**
Using an Xbox 360 controller works flawlessly with zero issues, but the adapter doesn't seem to detect the 8BitDo controller when wireless or wired (I believe this may in part be because the device ID changes between "8BitDo IDLE" and "8BitDo Ultimate C 2.4G Wireless Controller", despite the controller vibrating upon being plugged in, indicating a connection has been established. Included is a device descriptor dump, let me know if anything's missing or out of place.

**What type of controller do you have the issue with?**
8BitDo Ultimate C 2.4G Wireless Controller - 2dc8:3016 | 2dc8:3106

**What platform are you using the OGX-Mini on?**
OG Xbox + PC

**What board are you using?**
RP2040-Zero

[state1.txt](https://github.com/user-attachments/files/18217604/state1.txt)
[state2.txt](https://github.com/user-attachments/files/18217605/state2.txt)

mastercheff, 2024-12-24:
Same is true with its wired version as seen in #46 , doesn't work in 1.0 and didn't work with 0.4 either

USB\VID_2DC8&PID_3106&REV_0114
USB\VID_2DC8&PID_3106

While I didn't had a chance to packet sniff with mine  (yet) , Here's a photo of its main chip 

[image]

wiredopposite, 2024-12-24:
Does this adapter/controller have different modes like xinput, switch, etc. like the 8bitdo bluetooth adapter?

mastercheff, 2024-12-24:
Windows (both Dinput/Xinput) Android SteamOS  and RaspberryPI , it doesn't do Switch afaik

mastercheff, 2024-12-24:
Okay we dumb . Ultimate C 2.4g needs  X + (Home) to be pressed when connecting. Wired controller needs X be pressed when plugging in, it's in the manual. Will test tomorrow

mastercheff, 2024-12-25:
Okay first of all, Hardware IDs seem the same between two controllers, I'm using wired version.

There's literally zero difference when pressing and holding X upon plugging it in (tested on Windows 10) , despite what manual states

First it identifies itself as standard 360 controller

[image]

controller then vibrates couple of times and automatically changes into "Ultimate C Wired Controller" with axis readings not dissimilar to other controllers in Dinput mode. Windows sees it as a HID controller , despite being initialized briefly for a 360 controller

[image]

I also tried several Firmware revisions, it acts the same between them

mastercheff, 2024-12-25:
There's more . I got a bluescreen when trying to connect it multiple times over, leading to steamxbox.sys . Turned out controller bugged out because of Steam. After uninstalling "Xbox Extended Feature Support Driver" it finally behaves as it should, X upon connecting forces XInput, B button forces HID mode and selected mode sticks upon disconnects, similarly to deadzone setting. There's no DInput mode.

Still doesn't work with OGX-Mini, either XInput or HID

mastercheff, 2025-01-22:
as of alpha-3 , still an issue

wiredopposite, 2025-01-22:
I just ordered one of these so I'll figure it out once I get it, the descriptors in XInput mode look like a 360 controller so I'm not so certain why it wouldn't work as is

faithvoid, 2025-04-19:
Update on this: I tried again with a generic RP2350-A board (using both @jocover & @barrenechea 's modified firmware) , and something weird happens now. When the controller is connected via a wired conenction, the controller sometimes works for 10 or so seconds before disconnecting, and will crash the Xbox if reconnected after it disconnects. Using the wireless adapter doesn't lead to either result, despite the adapter recognizing there's a controller connected (but the inputs don't seem to do anything). Every other controller I have (wired/wireless Xbox 360 controller, wired Xbox One controller) seems to work fine without ever crashing the system or disconnecting, though.

faithvoid, 2025-04-23:
Second update: My other 3 "SpotPear" RP2350-A boards arrived today, and I can replicate the exact same thing on all 3 of them. I don't have a working multimeter on me to verify that each board is feeding a stable 5V to the controller, but the fact that it drops the connection after the exact same amount of time each time I've attempted it leads me to think it may be a controller-specific issue, maybe something with how it negotiates with the console it's plugged into? But I could be wrong (also: the LED doesn't turn off at any point, so it seems the board isn't power-cycling, it just acts as if the controller's been physically disconnected from the RP2350-A). Also, these tests were done in XInput mode, as DInput mode doesn't work in any scenario.

jocover, 2025-04-26:
> Second update: My other 3 "SpotPear" RP2350-A boards arrived today, and I can replicate the exact same thing on all 3 of them. I don't have a working multimeter on me to verify that each board is [...]

I use 8BitDo Ultimate 2C Wired Controller and it works fine
this controller reports speeds up to 1000hz, so I added a 4ms delay after tuh_task
[https://1fichier.com/?7ujp1791r0sroxa97ys3](url)
this is my own xinput to ps4 project. developed on Waveshare RP2350-USB-A, you can test it

mastercheff, 2025-04-26:
this is known @jocover 2C works for me too, original Ultimate C doesn't

faithvoid, 2025-05-02:
Chiming in again to confirm that this does seem to only exclusively affect the original 8BitDo Ultimate C controllers, as I bought a pink 2C (Bluetooth, Switch version) controller today that works flawlessly over a wired connection.

crashbandicode, 2025-10-02:
Just commenting to say my 2c works fine after adding the vid and PID as a d-input device