wiredopposite/OGX-Mini #59: HORI Nintendo Switch HORIPAD Wired Controller - Not Working

faithvoid, 2024-12-21:
**Describe the bug**
The HORI Nintendo Switch HORIPAD Wired Controller is not detected by OGX-Mini, but works just fine if plugged into a Mayflash NS adapter plugged into the OGX-Mini. Attached is a device descriptor.

**What type of controller do you have the issue with?**
HORI Nintendo Switch HORIPAD Wired Controller - 0f0d:00c1

**What platform are you using the OGX-Mini on?**
OG Xbox + PC

**What board are you using?**
RP2040-Zero

[horipad.txt](https://github.com/user-attachments/files/18217673/horipad.txt)

wiredopposite, 2025-01-01:
I added those IDs to alpha 2 on the releases page, let me know if that's working for you

faithvoid, 2025-02-08:
Not sure if the same changes are present in alpha 3 (assuming so), but using alpha 3 the Horipad still doesn't work when connected directly. If the same changes weren't made to alpha 3, let me know and I'll try again with alpha 2.