wiredopposite/OGX-Mini #60: FuSa Gamepad (PSP Gamepad HID Software) - Not Working

faithvoid, 2024-12-21:
**Describe the bug**
FuSa Gamepad (a PSP homebrew software that makes the PSP present as an HID gamepad) is not detected by the OGX-Mini, despite establishing a connection on the software side. Attached is a device descriptor. Not a priority whatsoever, but would be nice to have support for small party games or whatever.

**What type of controller do you have the issue with?**
FuSa Gamepad (PSP) - 00ff:01cb

**What platform are you using the OGX-Mini on?**
OG Xbox + PC

**What board are you using?**
RP2040-Zero

[fusa.txt](https://github.com/user-attachments/files/18217638/fusa.txt)

Invictaz, 2024-12-21:
Great. There is also homebrew like this for PSVita. I hope I don't forget to add this as well.

faithvoid, 2024-12-23:
Additional bug report: I played around with the HardwareIDs.h file to add the HID/VID of the FuSa Gamepad to it, and it "works" in the sense that it's detected, but it wildly spams inputs (I think namely the down input?). Buttons on the controller DO seem to do something while it's spamming inputs, so there's probably some weird handling issue.

wiredopposite, 2024-12-31:
Is there somewhere I can look at the source code for that project? I wouldn't really know which host driver to try and use for it

Invictaz, 2025-01-07:
> Additional bug report: I played around with the HardwareIDs.h file to add the HID/VID of the FuSa Gamepad to it, and it "works" in the sense that it's detected, but it wildly spams inputs (I think [...]

See this https://foosa.do.am/forum/6-180-2210-16-1368319869
Try the v0.1

Invictaz, 2025-01-08:
@wiredopposite here is the source 

https://github.com/andy-man/psp-fusa-gamepad