wiredopposite/OGX-Mini #64: Xbox 360 with UsbdSecPatch and Pico W Bluetooth Controller

domPL1, 2024-12-23:
With latest build after successfully connected Xbox Series X Controller (which I tested on PC) no input is recored in Xinput mode. Console does register that controller is pluged in (light on the front of console shows controller 1 is connected.

Controller
Xbox Series X Controller with latest firmware

Platform
 Xbox 360 RGH 3 With UsbdSecPatch

Board
Pi Pico W

Invictaz, 2024-12-23:
Try wireless. Usb on Pico W isn't implemented yet. See #56 

For now you can flash the Pico firmware to Pico W.
Then usb works

domPL1, 2024-12-23:
Sorry for not specifying. I paired Xbox controller to pico w (and it worked correctly with PC, but although Xbox showed connected no input was working). For now I reflashed older version and I use usb, but will reflash with latest non W version and let you know if that works  correctly over usb.

domPL1, 2024-12-23:
Pico non W version through usb is same. Console detects controller but no input is working. Reverted back to older version

Invictaz, 2024-12-23:
Which Series X firmware is it @domPL1 I will try with my series X controller to confirm.

domPL1, 2024-12-23:
Controller firmware 5.22.16.0

Invictaz, 2024-12-23:
> Controller firmware 5.22.16.0

[image]

Tested fine in 5.20.7.0 with OGX Mini v1.0. 
Honestly I don't dare to update since it might break and you cannot (easily) downgrade.

domPL1, 2024-12-23:
Maybe I will restate my issue. Between OGX Mini 0.4.0 and OGX Mini 1.0.0 support for Xbox 360 with UsbdSecPatch broke. Controller shows as detected on console, but non of button works.

purplebar0, 2024-12-23:
> Maybe I will restate my issue. Between OGX Mini 0.4.0 and OGX Mini 1.0.0 support for Xbox 360 with UsbdSecPatch broke. Controller shows as detected on console, but non of button works.

I can confirm this behaviour. No matter how many times I press Home or any other button, nothing happens, whereas during same boot, connecting a board with 0.4.0 (basic Pico, to be exact), it works as expected.

mrdude2478, 2024-12-23:
> > Controller firmware 5.22.16.0
> 
> ![image](https://private-user-images.githubusercontent.com/20338545/398221529-0475a716-fb57-43f3-84d8-93d1b2a7c447.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ5ODQ2ODQsIm5iZiI6MTczNDk4NDM4NCwicGF0aCI6Ii8yMDMzODU0NS8zOTgyMjE1MjktMDQ3NWE3MTYtZmI1Ny00M2YzLTg0ZDgtOTNkMWIyYTdjNDQ3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjIzVDIwMDYyNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWUyYmNiMjEyNDE3ZDViNjQ2ZTg4ODkxOTliMTk3NmE0OGFlYmQyMDk2ZGU1NWEzZjMyODU2Zjg3NzRhMWMzYWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.g0P7ViEz2NxWMSZLGQbSXeykZU2I1Mj72nYyL4UeRw8)
> 
> Tested fine in 5.20.7.0 with OGX Mini v1.0. Honestly I don't dare to update since it might break and you cannot (easily) downgrade.
> 
> I tried with xboxaccessories://firmwareupdate?legacyDowngrade=true and Series controllers cannot be firmware reverted.
> 
> The new update 5.22 brings higher polling rate and thus quite some changes. Maybe there is something needed.

All you need to do is put a momentary switch between GPIO0 and GND, It's just a case of pressing the switch while plugging into a usb port and dragging a utf file onto the drive that shows up. It takes seconds to upgrade and downgrade.

Invictaz, 2024-12-23:
> > > Controller firmware 5.22.16.0
> > 
> > 
> > ![image](https://private-user-images.githubusercontent.com/20338545/398221529-0475a716-fb57-43f3-84d8-93d1b2a7c447.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ5ODQ2ODQsIm5iZiI6MTczNDk4NDM4NCwicGF0aCI6Ii8yMDMzODU0NS8zOTgyMjE1MjktMDQ3NWE3MTYtZmI1Ny00M2YzLTg0ZDgtOTNkMWIyYTdjNDQ3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjIzVDIwMDYyNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWUyYmNiMjEyNDE3ZDViNjQ2ZTg4ODkxOTliMTk3NmE0OGFlYmQyMDk2ZGU1NWEzZjMyODU2Zjg3NzRhMWMzYWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.g0P7ViEz2NxWMSZLGQbSXeykZU2I1Mj72nYyL4UeRw8)
> > Tested fine in 5.20.7.0 with OGX Mini v1.0. Honestly I don't dare to update since it might break and you cannot (easily) downgrade.
> > I tried with xboxaccessories://firmwareupdate?legacyDowngrade=true and Series controllers cannot be firmware reverted.
> > The new update 5.22 brings higher polling rate and thus quite some changes. Maybe there is something needed.
> 
> All you need to do is put a momentary switch between GPIO0 and GND (if you don't already have a boot select button). It's just a case of pressing the switch while plugging into a usb port and dragging a utf file onto the drive that shows up. It takes seconds to upgrade and downgrade.

I mean the Xbox controller firmware which cannot be downgraded

mastercheff, 2024-12-24:
M$ playing dirty here lel

wiredopposite, 2024-12-24:
> Maybe I will restate my issue. Between OGX Mini 0.4.0 and OGX Mini 1.0.0 support for Xbox 360 with UsbdSecPatch broke. Controller shows as detected on console, but non of button works.

Looking into and fixing this is priority #1 when I get back from holiday

wiredopposite, 2025-01-01:
This should be fixed in alpha 2 on the releases page, let me know if it's working for you

purplebar0, 2025-01-01:
> This should be fixed in alpha 2 on the releases page, let me know if it's working for you

From a short amount of testing, it connects without an issue like the previous versions!

domPL1, 2025-01-02:
Yes, works correctly. Thank you very much. I will close issue

closed 2025-01-02 by domPL1