wiredopposite/OGX-Mini #65: (Fake) PowerA wired Switch controller / OpenSteamController

purplebar0, 2024-12-24:
**Is your feature request related to a problem? Please describe.**
Lack of a VID:PID corresponding to a controller that presumably works just like ordinary Switch controller.

**Describe the solution you'd like**
Simply a `20d6:a711` VID:PID addition.

**Additional context**
While this would likely make that original controller work, my motivation is to try out [OpenSteamController Wired Switch Pro Controller Mod](https://github.com/MichaelZaugg/OpenSteamController-Continued/tree/master/Mods/SwitchProWiredController), which makes Steam Controller emulate a wired Switch Pro Controller.

wiredopposite, 2024-12-31:
Which board are you using? I'll add the ids and upload a binary here for you to see if it works

purplebar0, 2025-01-01:
Board? I can test on a Pico and Pico W, if that's what you're asking.

As for whether it's the Switch Wired Controller or Switch Pro Controller, the name of the firmware suggests it emulates a Pro Controller.