wiredopposite/OGX-Mini #68: One PS3 Game Incompatibility

IdeactiveG, 2024-12-29:
**Describe the bug**
I've tried this with 8 PS3 games but for some reason GTA IV and the episodes version doesn't seem to like this mod which is a real shame as I like how it now has rumble support and the trigger sensitivity, well anyway back in version 0.4.0 the only buttons which didn't seem to work were up and down but they did work in the pause menu and with the new version they do work but certain activities like the in game phone and choosing which game to load into use X as circle and circle as square and I've tried to get the triggers to work but they wouldn't

**What type of controller do you have the issue with?**
HID\VID_054C&PID_0268
DualSense Controller

**What platform are you using the OGX-Mini on?**
PS3

**What board are you using?**
Pi Pico

wiredopposite, 2024-12-31:
Are the buttons switched only for some things and not others? I'll try to get ahold of a copy of GTA 4 and check this out

IdeactiveG, 2025-01-01:
yes certain in game things like the phone do work now in alpha1 but when trying to press x nothing happens but circle replaces x and to back out is circle but that was replaced by square and for some reason I couldn't get the triggers to work I have tried this with Max Payne 3 and it works a treat no issues I can recall, if you would like I could send you some videos with both 0.4.0 and alpha1 with GTA IV and its DLCs, thank you and I hope you have a happy new year

Invictaz, 2025-01-01:
@IdeactiveG try Alpha 2 now

IdeactiveG, 2025-01-01:
sorry but its still got the same issues