wiredopposite/OGX-Mini #71: 8bitdo Ultimate 2c not working on Pi pico/ Pi pico

Rm247, 2025-01-02:
Doesn't work on a pi pico using the usb port, also tried it in bluetooth mode on a pi pico w but without any luck, on pc it shows up in device menager when i connect the controller.

8bitdo Ultimate 2c wireless
HID\VID_2DC8&PID_310A&IG_01
HID\VID_2DC8&UP:0001_U:0005

Tried it on a modded xbox 360 with the patch installed and on pc, 

**What board are you using?**
Pi Pico, Pi pico W

mastercheff, 2025-01-02:
I have the same controller, and mine works in both modes (also works with the latest release). Make sure Your pico 40 pin outputs 5 Volts, some boards are underpowered and give problems especially with wireless dongles, but there are workarounds for that. Multimeter/Voltmeter would be the first step to diagnose further

Rm247, 2025-01-03:
Just checked it with a multimeter, shows 4.94 volts so i assume that shouldn't be an issue, the weird thing is that after plugging the controller into the pico it shows up as an unrecognized device in device menager, still no luck with even parying it to the web app, even after pressing the button combination on the controller.

mastercheff, 2025-01-03:
Okay, my firmware is 1.06 (factory firmware) I didn't dare to update it to beta 1.07 . Worked in every version of OGXMini that I tried.

Voltage seems Okay, but if it's stuck in some unrecognized state , I'd try to wake it somehow (FW update) so it works with Windows first. If it works with windows it should also work with ogxM there's no other way around it

Rm247, 2025-01-03:
Okay, so by some miracle after unplugging it and plugging it in countless times it randomly switched to xinput mode (didn't really change or press anything so i don't know why) now it works flawlessly on my xbox 360 wich is all i wanted it to do in the first place so i will leave it at that, thanks for the suggestions and your time, cant really tell what fixed it because i didn't even get to updating the controllers firmware like you said (it's still on 1.06)

closed 2025-01-03 by Rm247

mastercheff, 2025-01-04:
glad to help, Enjoy !