wiredopposite/OGX-Mini #76: RP2040 Won't work with newer fw than 0.40

Jean-DrEaD, 2025-01-09:
**Describe the bug**
I don't know if is a bug or just not understand correctly, but...
Tried ESP32 with RP2040-Zero and just won't detect... Downgraded one by one until work, I've not tested with ESP32 after get working(OGX-Mini_RP2040-Zero from 0.40 release).

**What type of controller do you have the issue with?**
Xbox360 and Xbox Series

**What platform are you using the OGX-Mini on?**
 OG Xbox, PC

**What board are you using?**
RP2040-Zero from Aliexpress

wiredopposite, 2025-01-11:
This issue is being addressed in the next update, you can get around it by pressing the reset button with your controller already plugged into the adapter, it should mount after that

wiredopposite, 2025-01-21:
Try out alpha3, it should be resolved now: [https://github.com/wiredopposite/OGX-Mini/releases/tag/v1.0.0-alpha3](https://github.com/wiredopposite/OGX-Mini/releases/tag/v1.0.0-alpha3)

wiredopposite, 2025-01-22:
This may be a different issue than the one I fixed in alpha 3, I'm talking about it with people in the discussions tab