wiredopposite/OGX-Mini #83: Web App Firmware Issue w/ RP2040-Zero

AnimeKid97, 2025-01-30:
**Describe the bug**
The web app firmware won't detect the usb port. It detects the board, just not the usb port. Also, I'm not sure if this is normal but it doesn't seem to finish installing the firmware. The first update works fine, but the 2nd update just hangs every time. The alpha3 firmware works fine, and I can still reinstall it.

**What type of controller do you have the issue with?**
Any controller

**What platform are you using the OGX-Mini on?**
 OG Xbox

**What board are you using?**
OGX-Mini plug and play v2 RP2040-Zero (Green PCB) by ModsvilleUSA

wiredopposite, 2025-01-30:
The firmware updater is for the wireless Pico+ESP32 board, not for any wired adapters, I'll update the page so maybe it'll be less confusing. I'm not sure what firmware modzville used, I'd assume it's the RP_ZERO or PI_PICO firmware, but update your adapter with one of those from the **[Releases page](https://github.com/wiredopposite/OGX-Mini/releases/tag/v1.0.0-alpha3)**, you might have to ask him what he's using. Then plug in a controller and hold Start+LB+RB to get into web app mode, the web app should be able to connect over USB after that.

AnimeKid97, 2025-01-30:
Ok, I took your word for it. It does pair up with the alpha3 firmware. But, I came across another problem. Switch pro controller seems to pair up with a Bluetooth adapter, and works fine connected to an OG Xbox. But when it's just plugged straight in without the Bluetooth adapter, my character is slow when holding right. The controller doesn't want to turn on either when I try to pair it wired with the web app. It charges, but it can't turn on. I also tested a PS4 controller and it works perfectly with and without the Bluetooth adapter. Button remapping as well also works.