wiredopposite/OGX-Mini #87: Pico W PS3 output problem

Vitari-Translates, 2025-02-01:
**Describe the bug**
The adapter is malfunctioning in PS3 output mode. In the main menu everything is fine. I tried Ratchet&Clank there it seemed to work, but for some reason the adapter sends strange input. For example the character starts running backwards. It does this without a controller connected, and the PS3 keeps saying that the controller is low on charge. If I start using the controller then everything works, but the error does not go away.  Sly 2 and Jak 2 don't even detect the adapter.
I have a Pi Pico W.

**What type of controller do you have the issue with?**
Any controller

**What platform are you using the OGX-Mini on?**
PS3

**What board are you using?**
Pi Pico W

Invictaz, 2025-03-15:
See #3

TRRAXT, 2025-03-18:
> See [#3](https://github.com/wiredopposite/OGX-Mini/issues/3) and
> 
> [#16 (comment)](https://github.com/wiredopposite/OGX-Mini/issues/16#issuecomment-2083900931)
> 
> and
> 
> [#16 (comment)](https://github.com/wiredopposite/OGX-Mini/issues/16#issuecomment-2585398813)
> 
> Because the descriptor name is not given and the output is not setup correctly this behavior might happen.

Are you suggesting that "If no proper string is present on the Switch and PS Vita will not be able to have it working correctly." could this be the reason?
Can you fork or make a commit?