wiredopposite/OGX-Mini #88: Question: Nintendo Switch mode and rumble

TRRAXT, 2025-02-01:
1. Using Nintendo Switch output mode and Xbox 360 (Fat) RF board made as receiver, but doesn't work rumble.
Is OGX-Mini has support for rumble? or it just my receiver/gamepad faulty?

2. In Nintendo Switch output mode OGX-Mini on Switch shows as which controller?
if not as pro con, can it be changed to pro con?

wiredopposite, thank you so much for this project!

TRRAXT, 2025-02-02:
upd:
1. Using Xbox Series controller can't get rumble.

2. OGX-Mini shows as some of Hori controller.
Can be added switch/toggle in web app to switch between pro con or Hori?

TRRAXT, 2025-02-07:
2. was covered in pull request https://github.com/wiredopposite/OGX-Mini/pull/61

jretropie, 2025-07-01:
I have this same issue with Xbox 360 controller + RF board receiver.  Rumble does not work on switch console.