wiredopposite/OGX-Mini #91: Any way to get an only ESP32-S3 plugged to console to use wireless only controller ?

Benten1000000, 2025-02-04:
**Is your feature request related to a problem? Please describe.**
Not really but something simplier

**Describe the solution you'd like**
I would like to use a simple setup with only an OG Xbox Controller cable to USB-C to ESP32-S3

**Additional context**
----

hakanrw, 2025-02-15:
@wiredopposite did you encounter a technical problem that prevents ESP32 being the sole component of the gamepad emulator, without the Pico?

i'm interested in this project, and i've been inspecting the code for a while. i would be happy to be of help.

wiredopposite, 2025-03-31:
Hey, no there's no issue with getting the S3 working, it just doesn't support bluetooth classic so the only things that would really work on there are newer Xbox controllers. I haven't found any mcus that support bt classic and have built in usb, if you know of one I'd definitely make some firmware for that.