wiredopposite/OGX-Mini #93: PS3 mode not working properly

TRRAXT, 2025-02-05:
**Describe the bug**
On PS3 console in some games working fine, in some (as NFS:Most Wanted) wierd/phantom input.
On PC with official Sony driver for Dualshock 3 it has strange + phantom input.
Some buttons pressed, some not working and sticks work as one stick.
This can be problem with driver itself, but in some PS3 games (on stock console) I have wierd/phantom input.

[image]
In a free state

Not working: LB, RB, RT, A, B, X, Guide.
Working: Left stick, LT, L3, R3, Dpad UP, Dpad Right, Dpad Down.
Working strange: Right stick works only horizontaly (and recognized as left stick?), Dpad Left changes state not only button, but also one of axis, X changes state of axis (not button).

**What type of controller do you have the issue with?**
Xbox Series controller and Xbox 360 Controller with reciever.

**What platform are you using the OGX-Mini on?**
PS3, Switch.

**What board are you using?**
RP2040-Zero

Invictaz, 2025-03-15:
I posted some suggestions in #87

wiredopposite, 2025-03-31:
I think this had to do with trying to push USB reports too frequently, I'm gonna do some more testing on the new update before posting the release but from my initial testing it seems fixed

lopenator, 2025-09-18:
Happening to me also.