wiredopposite/OGX-Mini #94: Xbox 360 Wireless Adapter for all 4 controller ports

dazzaXx, 2025-02-13:
**Is your feature request related to a problem? Please describe.**
I would like to use an Xbox 360 Wireless Adapter for all 4 controller ports, as using the one pi only works for the one player it's plugged into.

**Describe the solution you'd like**
Is it possible to link 4 pi's together similarly to how the ogx360 did with 4 arduino's with your firmware?

dazzaXx, 2025-02-13:
Apologies, I see it does support this, is there any quick steps that could tell me how to link several pi's together? I see that it is similar to how the ogx360 handles but wanted to double check, thanks.

romancherednikovopsfleet, 2025-02-20:
@dazzaXx https://github.com/wiredopposite/OGX-Mini/issues/72

dazzaXx, 2025-02-20:
> [@dazzaXx](https://github.com/dazzaXx) [#72](https://github.com/wiredopposite/OGX-Mini/issues/72)

Thank you. I'll close this issue as I didn't see there was one already open. Apologies.

closed 2025-02-20 by dazzaXx