wiredopposite/OGX-Mini #95: Xbox button 'sticky'

Luke7352, 2025-02-15:
Xbox button behaves 'sticky' on rp2040-zero with xbox elite and PDP xbox controller

Xbox Elite 2/PDP Controller

Xbox360/PC Xinput

RP2040-Zero

When i press the xbox button, the buttons for pressing in the left stick, the left bumper and the xbox button bind, which resets the next time you input something. No other buttons behave this way.

Luke7352, 2025-02-15:
Just quickly wires up to my breadboard and i get the same issue on my pico

Luke7352, 2025-02-15:
this issue was introduced in 1.0a1, however my probklem is not solved as 0.4.0 does not support my pdp xbox controller. only my elite controller

Luke7352, 2025-02-15:
https://github.com/user-attachments/assets/ab6cdc69-c6a1-4928-aeb2-73e4712b8a48

This is the behaviour

EatPrilosec, 2025-04-05:
i am also experiencing this. pressing any button after the guide button seems to unstick it

Luke7352, 2025-04-05:
> i am also experiencing this. pressing any button after the guide button seems to unstick it
> 
> happens on every controller ive tried:
> 
>     * dualshock4 (wired and wireless)
> 
>     * same controller as OP
> 
>     * hyperkin duke 20th an.
> 
>     * 3rd party xbox360 controller clone
> 
> 
> and im using an rp2040 zero

Exactly this!!

AKeilhauer, 2025-04-05:
Can confirm, happens on all of my Original Xbox One Controllers (Micro-USB and USB-C Version)
Did not happen on my Original Xbox 360 Wired Controller.
Tried all firmwares down to 0.4 on a pico1

faithvoid, 2025-04-22:
Can also confirm this behaviour on both the RP2040 and RP2350-A with an Xbox 360 and Xbox Series controller.

belps96, 2025-09-01:
Same issue here on a Pico with an Xbox One controller