wiredopposite/OGX-Mini #97: Swap bumpers and triggers

Luke7352, 2025-02-19:
**Is your feature request related to a problem? Please describe.*
Allow buttons to be mapped to triggers, eg. pressing lb on a controller will act as lt

**Describe the solution you'd like**
Add l2/r2/lt/rt to options for what a button can be remapped to

Invictaz, 2025-03-29:
Maybe the use of Joycon Driver can help to have reverse sticks

https://github.com/fossephate/JoyCon-Driver

Luke7352, 2025-03-29:
> Maybe the use of Joycon Driver can help to have reverse sticks
> 
> https://github.com/fossephate/JoyCon-Driver

I am attempting to use this on an Xbox 360

mastercheff, 2025-04-11:
[request] in the name of the issue would be helpful. otherwise, Bump!

Luke7352, 2025-04-12:
> [request] in the name of the issue would be helpful. otherwise, Bump!

Sorry I don't do this often have added it