wiredopposite/OGX-Mini #110: DualShock 4 PS button not working

Lyall-A, 2025-04-26:
**Describe the bug**
The PS button isn't getting registered on both PC and Xbox 360, work's when directly plugged into PC without the Pico

**What type of controller do you have the issue with?**
DualShock 4

**What platform are you using the OGX-Mini on?**
Xbox 360

**What board are you using?**
Pi Pico (RP2040)

Lyall-A, 2025-05-14:
I've connected up a ESP32 for Bluetooth and it work's find over Bluetooth (at least on PC anyway)

Adro8, 2025-05-22:
I'm using a rp2040zero and I have similar issue, my ps button doesn't get recognized on my switch.

dee098, 2025-05-23:
Same problem DS4 wired controller PS button on and PlayStation 3 system
(FW - OGX-Mini-v1.0.0a3-PI_PICO.uf2)

inhellcorruption, 2025-06-28:
Same problem with my DS4 v1, PS button not working on Xbox 360