wiredopposite/OGX-Mini #117: [Feature Request] libxsm3 implementation

Loomeh, 2025-05-08:
The process of emulating an XInput controller on Xbox 360 currently requires hardmodding your console via JTAG or RGH and installing [UsbdSecPatch](https://github.com/InvoxiPlayGames/UsbdSecPatch) to bypass Xbox Security Method 3 controller authentication. An implementation of [libxsm3](https://github.com/InvoxiPlayGames/libxsm3) would make installing the patch unnecessary and would allow OGX-Mini to work on stock/retail consoles.

baabo, 2025-05-08:
I came here to request the same as the retail keys were recently discovered. Would be a great addition to the project.

Invictaz, 2025-05-10:
It's now compatible with the Pico. But please keep this in the #15 thread