wiredopposite/OGX-Mini #125: My Pi Pico board fried ? Did it happen to someone else ?

LeGittee, 2025-05-24:
Hello,

First of all thank you for making this project, I am looking for a way to play with my Logitech F710 on an unmodded Xbox 360 and OGX mini has allowed me to do it for a very short time, in combination of a cronus adapter (which does not support the F710 by default)

So like I said I used this program on a regular Raspberry Pi Pico board through a Cronus Max to play on an Xbox 360 with a F710 controller. It worked for about 30 minutes, the inputs were fine but I think the rumbling was not 100% functional. Then it seems to have broke, the inputs were not working and my controller was rumbling constantly until I unplug the receiver.

I plugged the gamepad receiver through a USB breakout board that I soldered to a USB cable which is soldered to GP0 and  GP1 for D+ and D-. 
Since the board is detected by my PC, I assume it is the GPIO ports that are fried.

I want to mention that I previously had the same issue with "HID remapper" on another Pico board.

Since I assume this is a hardware issue, I don't expect a software solution for it but I was curious to know if it had happened to somebody else before ?

LeGittee, 2025-05-26:
I used a RP2040, I bought it from Aliexpress, so I don't know if it really official, but it seems to be.
The breakout board is soldered correctly. Although the green wire on the pico board end (D+) Is kind of exposed and quite close to the micro USB connector shield, but in case of a short to the ground it should not fry the GPIO port ?

Here is a picture:
[image]

Some of the usb wire shield is visible but it does not touch anything, I was just too lazy to cut it

mastercheff, 2025-05-26:
Looks like super sloppy job to me. Shielding wires loose is literally asking for a short somewhere...

Re-do Your soldering and use a dab of flux/rosin ffs

faithvoid, 2025-05-27:
I'm not a soldering expert, but I agree with @mastercheff 's assessment, those shield wires shouldn't just be dangling like that and in my opinion there's slightly too much exposed wire at the solder points on the Pico (the red and black wires looks barely connected from what I can see). Also be sure to use flux if you weren't already. 

Don't be fully discouraged, I wouldn't consider the board dead until you've done a full re-solder job and tested everything again as it could just be a loose connection, but if you're just looking for something that works and don't trust your soldering abilities, you can buy an RP2350-A (the board with USB-C and USB-A female ports built-in) from AliExpress or similar for approximately $10 CAD per board and use barrenechea's OGX-Mini fork in the interim.

LeGittee, 2025-05-27:
I was already using rosin flux and there is continuity on all 4 wires. Did you try OGX mini with the RP2350-A yourself ?

mastercheff, 2025-05-27:
I did one with the RP2350A, it's flawless with barrenechea's forked firmware