wiredopposite/OGX-Mini #132: How does Bluetooth with this work?

Caitlynmcc234, 2025-05-31:
I want to make this so I can use my Xbox series controller wirelessly with my switch. Is this project capable of doing that?

Invictaz, 2025-05-31:
Yes. Load the Pico W firmware onto a Pico W.

Dtpk, 2025-06-05:
For mine I used the esp32 + pico build I cant say nothing about the pico w but looks like that version gets more support so use that....

but the 2 board combo I am only able to use my xbox one remotes over bluetooth my wii remotes and ps4 remotes refuse to connect. 

- But from just using my xbox remotes over it they been working fine - 

although the guide button does nothing on the xbox 360 for me at least kinda makes it hard to you know quit a game hope they can add a combo or fix that.

Sometimes rumble seems to be delayed which is another weird thing.